MKFirmware/MK4duo #620: Too many compilation warnings on GT2560

siana, 2018-11-09:
To reproduce set #define MOTHERBOARD BOARD_GT2560_REV_A_PLUS
and #define SDSUPPORT

Following warning is repeated numerous times, it's nonfatal but super annoying:

In file included from sketch\src\platform\HAL_SAMD\../../../src/inc/pins.h:28:0,

                 from sketch\src\platform\HAL_SAMD\../../../MK4duo.h:105,

                 from sketch\src\platform\HAL_SAMD\HAL_timers.cpp:53:

sketch\src\platform\HAL_SAMD\../../../src/inc/../boards/75.h:186:0: warning: "SD_DETECT_PIN" redefined

     #define SD_DETECT_PIN     38

MagoKimbra, 2018-11-09:
You did not set up the right processor on Arduino IDE! He is compiling for SAMD.

siana, 2018-11-09:
I decidedly have the right processor selected! You're pulling in all the HALs somewhere.

[image]

Besides the warning is in all compilation units rather than one, so it reoccurs dozens of times.

If you look in 75.h, you can clearly see SD_DETECT pin being defined
#define SD_DETECT_PIN              NoPin
in line 150
and then #if ENABLED(ULTRA_LCD)
it's redefined a few times later without being #undef 

I inserted #undef SD_DETECT_PIN at line 171 to remove the warnings.

siana, 2018-11-09:
[complog.txt](https://github.com/MKFirmware/MK4duo/files/2566912/complog.txt)

MagoKimbra, 2018-11-09:
The compiler warnings must be put on disabled ...

siana, 2018-11-09:
There's no reason to disable warnings. It's not that hard to have a clean preprocessor instruction flow that doesn't have conflicting definitions.

siana, 2018-11-10:
Note: the log file above was generated with setting Preferences -> Settings -> Compiler Warnings: None.

MagoKimbra, 2018-11-10:
This is a bug the Arduino IDE.. Now open arduino ide and set compiler warning to another value.
Close arduino IDE and reopen and set None.
Closed and reopen and now the compiler warning is set to None...

closed 2018-11-15 by iosonopersia