MKFirmware/MK4duo #621: Watchdog causes EEPROM Reset to hang

siana, 2018-11-09:
If #define USE_WATCHDOG is set, selecting clear EEPROM or restore failsafe from the menu causes the firmware to hang, with the board LED flashing a few times a second.

WATCHDOG_RESET_MANUAL is not set. The processor is ATMega2560.

The corresponding feature set works in mainline Marlin.

In general i am quite disappointed by the safety approach of this firmware. Thermal runaway disabled by default, watchdog disabled and broken. Our machines pose a fire danger and every little bit the firmware can do to keep them secure is necessary. The watchdog is not only there to guard against firmware bugs, but against hardware malfunction too.

MagoKimbra, 2018-11-09:
The firmware has many other things for safety, such as the automatic shutdown of all the heaters when it detects false temperatures or after 30 minutes that it does not print. The thermal runways are not enabled by default because they are advanced functions that the user must define and set. I do not see where the problem is for not having default settings active.
As for the watch dog I have enabled both on my delta with a 32bit card on the wanaho D6 with 8-bit card and I have no problem. I do not understand what happens to you ...

siana, 2018-11-10:
Resolved by 064c8e9.

closed 2018-11-10 by siana