MKFirmware/MK4duo #623: won't compile using BOARD_MINITRONICS_V2

ivanlunardi, 2018-11-09:
ketch/src/platform/common/communication.cpp.o: In function `Com::serialFlush()':
sketch/src/platform/common/communication.cpp:65: undefined reference to `Serial'
sketch/src/platform/common/communication.cpp.o: In function `Com::serialRead(unsigned char)':
sketch/src/platform/common/communication.cpp:75: undefined reference to `Serial'
sketch/src/platform/common/communication.cpp.o: In function `Com::serialDataAvailable()':
sketch/src/platform/common/communication.cpp:83: undefined reference to `Serial'
sketch/src/platform/common/communication.cpp.o: In function `Com::printPGM(char const*)':
sketch/src/platform/common/communication.cpp:103: undefined reference to `Serial'
sketch/src/platform/common/communication.cpp.o: In function `Com::write(char)':
sketch/src/platform/common/communication.cpp:110: undefined reference to `Serial'
sketch/src/platform/common/communication.cpp.o:sketch/src/platform/common/communication.cpp:58: more undefined references to `Serial' follow
collect2: error: ld returned 1 exit status
exit status 1
Errore durante la compilazione per la scheda Minitronics v2.0.

MagoKimbra, 2018-11-09:
For the minitronics v2 we are still working together with RRW. For now the firmware is not yet final.

brupje, 2018-11-18:
You are using 1.0.1 of the ReprapWorld SAMD board definition?

tturpin, 2019-11-22:
Hi,
What is the status of Minitronics v2.0 support ?
I can successfully compile and use V4_3_8 for with the Minitronics v2.0 board, but if I try either V4_3_9 or master, I get the following compilation error:

```
In file included from /home/tifn/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/samd21/include/samd21.h:83:0,
                 from /home/tifn/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/samd.h:117,
                 from /home/tifn/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/sam.h:470,
                 from /home/tifn/.arduino15/packages/ReprapWorld/hardware/samd/1.0.2/cores/arduino/Arduino.h:48,
                 from sketch/MK4duo.ino.cpp:1:
/home/tifn/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/samd21/include/samd21j18a.h:488:38: error: expected ')' before '*' token
 #define PORT              ((Port     *)0x41004400UL) /**< \brief (PORT) APB Base Address */
                                      ^
sketch/src/platform/common/serial.h:80:24: note: in expansion of macro 'PORT'
   static constexpr int PORT               = serial;
```
I tried renaming "static constexpr int PORT" in platform/common/serial.h but then I get other errors:
```
sketch/src/platform/HAL_SAMD/HAL.cpp: In function 'void tone(pin_t, uint16_t, uint16_t)':
sketch/src/platform/HAL_SAMD/HAL.cpp:106:35: error: invalid conversion from 'millis_s {aka short unsigned int}' to 'millis_s* {aka short unsigned int*}' [-fpermissive]
   while (pending(endTime, duration)) {
                                   ^
In file included from sketch/src/platform/HAL_SAMD/../../../MK4duo.h:115:0,
                 from sketch/src/platform/HAL_SAMD/HAL.cpp:55:
sketch/src/platform/HAL_SAMD/../../../src/utility/utility.h:30:19: error:   initializing argument 1 of 'bool pending(millis_s*, millis_s)' [-fpermissive]
 FORCE_INLINE bool pending(millis_s *start, const millis_s period) { return !expired(start, period); }
                   ^
```
with master, or
```
In file included from sketch/src/core/printer/../../../src/platform/platform.h:34:0,
                 from sketch/src/core/printer/../../../MK4duo.h:109,
                 from sketch/src/core/printer/printer.cpp:29:
sketch/src/core/printer/printer.cpp: In static member function 'static void Printer::setup()':
sketch/src/core/printer/printer.cpp:165:48: error: 'freeMemory' was not declared in this scope
   SERIAL_SMV(ECHO, MSG_FREE_MEMORY, freeMemory());

                                                ^
```
with V4_3_9.

So, I'm stuck with V_4_3_8, but the sdcard writing (M28) does not work (I suspect the file opening function is broken, and this seems to have completely changed in later versions). Any suggestions ?

Diexon, 2020-04-24:
Hi,

I recently purchased a Minitronics 2.0 and I am stuck with the same problem. Will this board not be supported any longer?