MKFirmware/MK4duo #624: Saving printer statistic in EEPROM

ZhenyaKa, 2018-11-09:
In original Marlin statistic saved in EEPROM.

In Kimbra realised saving in SDCARD only.

All Marlin code was cleared/rewritten.

Is it planned option to choose where to store statistics (EEPROM or SDCARD).

SD CARD may be different each time and no good solution to store stat there.

Also I have random printer hangings when saving data to SD CARD.

MagoKimbra, 2018-11-09:
I have statistics before Marlin, and I have chosen to put them on sd because the eeprom has a finite number of readings and writes. I prefer them to be on SD only.

closed 2018-12-28 by iosonopersia

ZhenyaKa, 2018-12-28:
I think that no need to cut functionality and users can choose themself.
Use eeprom or flash

MagoKimbra, 2018-12-28:
The statistics of the machine are saved continuously about every 5 minutes, in the eeprom it would be like destroying it, it would have a short life, it is not feasible. In the sd there is no problem because it is a moment that no problem with the printer.

ZhenyaKa, 2018-12-28:
24lc256 - eeprom chip for DUE. It has 256k

EEPROM settings for KIMBRA have max 1kb of data. You have 256 DIFFERENT PLACES to store/read values. And no any reasons to destroy EEPROM quickly!!!

Just save data each time in other region and thats all. Where is the PROBLEM here???

MagoKimbra, 2018-12-28:
But I can not change the position every time. The eeprom reads and writes with an address, so if I write with address 0x010 for example when I go to read I have to go to the address 0x010. When I turn on the printer the firmware how do you know the last used address?
So the position is always the same same cell for each parameter.

ZhenyaKa, 2018-12-29:
[http://ipic.su/img/img7/fs/kiss_16kb.1546071038.png](url)

Look at this EEPROM "Window" solution.

MagoKimbra, 2018-12-29:
This is in Marlin:
```
/**
  * @brief Interval in seconds between EEPROM saves
  * @details This const value defines what will be the time between each
  * EEPROM save cycle, the development team recommends to set this value
  * no lower than 3600 secs (1 hour).
  */
```
This means that it saves every hour, if you turn off the printer before it has saved you lose an hour of statistics.
I do not understand how do you find the maxid when the printer is turned on?

ZhenyaKa, 2018-12-29:
did you read text on my pic?

Max value easily get by one cycle.

MagoKimbra, 2018-12-29:
Yes i read, but i do not understand the max value where i found when i turn on printer??

ZhenyaKa, 2018-12-29:
read until got zero or less value in next cell of ID. Previous block of values will be actual block of data

MagoKimbra, 2018-12-29:
But the cells are not erased, I have to physically write 0 to delete them, so instead of writing once I have to write twice once for the value and once to delete the previous cell.

ZhenyaKa, 2018-12-29:
When you reach max ID value you will erase all ID and thats all

ZhenyaKa, 2018-12-29:
Max I mean really Max for that window size.

ZhenyaKa, 2018-12-31:
WOW THANX!