MKFirmware/MK4duo #625: Arduino Due software serial library

weed2all, 2018-11-10:
More like a question...i encountered on google someone that successfully made a software serial library for arduino due...my question is how can this be a added to the mk4duo in order that we(arduino due + radds users) can have our tmc2208 uart communication
Link for the library:

https://github.com/antodom/soft_uart

MagoKimbra, 2018-11-10:
The problem is that MK does not have to do it, but the library for the tmc2208 if they use the serial software then everything is ok.

weed2all, 2018-11-10:
Ohh...ok...seams that will not be so easy then...i don't have the knowledge require to do this...

MagoKimbra, 2018-11-10:
Sorry but my English is not perfect. I did not say that you have to do it, but the one who writes the library for TMC2208 drivers.

weed2all, 2018-11-10:
Yes would be awsome if it will do...i would love to have this working...is the last thing I want from this board...already have mos tft working,I'm on the go to receive my nextion 5" as well...but the nextion I know for sure it will work...so the remaining task is to have uart over my tmc2208 on radds+arduino due...i will be the happiest guy...

weed2all, 2018-11-29:
Hey...any news on this?i see the tmc library have support for software serial on tmc2208...my question is how to add the above software serial library into mk4duo for own use?im complete noob on this...but as i said i would love to have my tmc2208 uart on radds+due...

closed 2019-03-20 by weed2all