MKFirmware/MK4duo #634: 4.3.7 Nextion not detected

jgauchia, 2018-11-22:
After uploaded 4.3.7 Firmware and Nextion Firmware
LCD doesn't detected.
With previous version Nextion is detected

LCD was connect to Serial2 (pin 17,16) of RAMPS

Maybe it will need a redefinition pin?

MagoKimbra, 2018-11-27:
None, it works well on all cards!

jgauchia, 2018-11-27:
MMM ok, I'll try again.. :( txs!

MagoKimbra, 2018-11-27:
Version of your Nextion? 4.3 or 7" or??

jgauchia, 2018-11-27:
4.3"

zampollitr, 2018-12-01:
I confirm the problem: Nextion 4.3 is not detected on version 4.3.7. 
On version 4.3.6 works without problems.
LCD was connect to Serial 2 (pin 17.16) of RAMPS

zampollitr, 2018-12-01:
Um... created in the Configurator firmware and flashed to the printer-now Pronterface see Nextion, but then nothing happens.
Printer does not respond to the panel...
In the console Pronterface is a cyclic reconnection:
Connecting...
start
Printer is now online.
Info:echo:MK4duo_4.3.7
echo:Last Updated: Dec  1 2018 22:36:47 | Author: (none, default config)
echo:Compiled: Dec  1 2018
echo:Free Memory: 2216 PlannerBufferBytes: 1184
Nextion 4.3" connected!
start
Info:echo:MK4duo_4.3.7
echo:Last Updated: Dec  1 2018 22:36:47 | Author: (none, default config)
echo:Compiled: Dec  1 2018
echo:Free Memory: 2216 PlannerBufferBytes: 1184
Nextion 4.3" connected!

jgauchia, 2018-12-01:
It's clear, something happens

MagoKimbra, 2018-12-01:
Fixed connection Nextion with Arduino MEGA!

jgauchia, 2018-12-02:
Ok thanks, Tomorrow I'll download and try it

zampollitr, 2018-12-02:
Thanks, Mago! 4.3.7 works great!

jgauchia, 2018-12-03:
OK, Nextion is detected!!! (well, move options, etc.. doesn't works, well never worked for me.) 
I'll close the issu

closed 2018-12-03 by jgauchia

MagoKimbra, 2018-12-03:
@jgauchia What is not working?
What card do you have?
What Nextion do you have?

jgauchia, 2018-12-03:
Ramps with Atmega2560 + nextion 4.3" basic.
Options in menu like move doesn't works, if I make homming Z axis for example, printer resets and loops constantly until leave movement menu.
Gcode menu and set temp menu doesn't make anything.
Only speed menu works.
This occured too in other versions of Firmware , I think that I opened some issues about it (I don't remember now).
Really it isn't a "great" problem if doesn't works.

jgauchia, 2018-12-03:
**** EDIT ****
With nextion fw of this new release (updated now) temperature works

MagoKimbra, 2018-12-03:
ok no problem... I do not want the problems to stay, I want to solve them. I do not always succeed because the nextion on the mega I have no way to prove it.