MKFirmware/MK4duo #640: Power supply "1" or "2 "no running with version 3.4.7

Djac1974, 2018-12-04:
### Issue description
The compilation is ok, when I upload on arduino due and radd, the serial monitor doesn't change, doesn't evolve. This page is white, the automatic scrolling is activated, no evolution.
i suppose that  power supply 1 or 2 stop  a program's progression

### Compile errors
I don't have problem with compilation.

### Firmware version
MK4duo 3.4.7 No ok
with 3.4.6 (2018-may) no probem.

### My setup
[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/2643388/Configuration_Overall.txt)
[Configuration_Pins.txt](https://github.com/MKFirmware/MK4duo/files/2643389/Configuration_Pins.txt)

Djac1974, 2018-12-11:
Has anyone managed to use this feature? power supply ATX or power supply Box360 with version 3.4.7, it will work correctly in version 3.4.6 (2018-05-18)
Thanks in advance.

Djac1974, 2018-12-12:
Thanks marlinkimbra for the modification of yesterday

Now the PS-ON works, with the parameters:
#define POWER_SUPPLY 2
#define PS_DEFAULT_OFF false
but serial monitor's arduino remains white and the nextion remains on connect

if i change the configuration with:
#define POWER_SUPPLY 2
#define PS_DEFAULT_OFF true
then serial monitor works and the nextion also. but the PS-ON is not functional.

Djac1974, 2018-12-13:
I am testing the modification and I see that with:
#define POWER_SUPPLY 2
#define PS_DEFAULT_OFF true
PS-On is maintained only at the request of an action: heating of head, move axis, ....
While  version 4.3.6 on a pulse on the button "Start for exemple" PS-ON started immediately.

MagoKimbra, 2018-12-23:
Please try now... I check the difference with version 4.3.6 and i found only two difference.. But i am not sure if this difference fix the problem...

Djac1974, 2018-12-23:
Hello MagoKimbra, I tested the new version, the problem is the same.
With configuration:
#define POWER_SUPPLY 2
#define PS_DEFAULT_OFF false
I suspect a problem of exchanging data the screen which remains on connect and the serial monitor on a white screen.
I have attached the two overhall files 4.3.6 and 4.3.7
Thank you for your help.

[Configuration_Overall_4.3.6.txt](https://github.com/MKFirmware/MK4duo/files/2706303/Configuration_Overall_4.3.6.txt)
[Configuration_Overall_4.3.7.txt](https://github.com/MKFirmware/MK4duo/files/2706302/Configuration_Overall_4.3.7.txt)

MagoKimbra, 2018-12-23:
Excuse me, but I can not understand what the problem is. When you turn the machine on, the power supply is on and everything is working, but what's wrong?

Djac1974, 2019-03-01:
Hello, back after hospitalization,
The problem is as follows: it is necessary to maintain the "start push" until the request for changing the temperature of the bed, the heating head or the selection of an object to be printed.
On version 3.4.6 (2018-05-18), a "start push" was activated and all energies were activated.
On version 3.4.7 we no longer have this self-maintenance of K1-K2 on a pulse of "start push".

Djac1974, 2019-05-26:
Hello,
You can close this issue, with version 4.7.9 no problem.

Thank you.

commit 5e3ba02c (2018-12-23): Likely fix #640
closed 2019-05-27 by iosonopersia