MKFirmware/MK4duo #645: 4.3.6 and 4.3.7 on mks 1.4 bed heating bug.

svsdval, 2018-12-16:
I using mks 1.4 with mk4duo 3.4.5 and works fine, but on 3.4.6 and 3.4.7 I have problem with bed heating:
After use online configuration, I upload new firmware to the printer. In configuration menu I use bed auto pid heated to 60*C, (auto pid works fine - bed heater is works) when auto pid is passed i tried set bed temp manually to 100*C, on the screen firmware accepts new target temp to 100*C, but no changes - bed heater stay disabled. If run again bed auto pid to 100*C - works fine - bed heater is works...

[Configuration_Overall.h.346.zip](https://github.com/MKFirmware/MK4duo/files/2683490/Configuration_Overall.h.346.zip)

iosonopersia, 2018-12-16:
I think you need to save PID values in EEPROM (have you enabled it?) with M500. Firstly do the pid autotune, then send M500 and M501. Then you should be able to control the heater correctly. If this doesn't work, then there's a problem and we need to figure it out.

MagoKimbra, 2018-12-16:
If you do the pid with the M303 command without giving the U1 subcommand, the pids are only written at the end but they are not updated, so you have to give the command M301 P I D with the corresponding values and then M500.
If instead you give the command M303 H-1 U1 then the values found update those in memory and are saved directly in EEPROM.

svsdval, 2018-12-17:
> to save PID values in EEPROM (have you enabled it?)
yes, i saved the configuration, but not helped...

ok, i was found somethings ...
in serial port , in 3.4.7 configuration hotbed is missing, may be some bug in definition or compilation ?
[pid-3.4.7.txt](https://github.com/MKFirmware/MK4duo/files/2686104/pid-3.4.7.txt)
[pid-3.4.5.txt](https://github.com/MKFirmware/MK4duo/files/2686105/pid-3.4.5.txt)

Ratisse, 2019-02-01:
Had same issue, solved after clearing the MKS 1.4 eeprom via Arduino IDE using this code: https://www.arduino.cc/en/Tutorial/EEPROMClear , then re upload MK4duo firmware

Autotune PID worked fine but not saved in eeprom, now it's OK

svsdval, 2019-02-02:
not helped to me ;( tried on 4.3.7 and 4.3.8 )
[4.3.7.log.txt](https://github.com/MKFirmware/MK4duo/files/2824154/4.3.7.log.txt)

MagoKimbra, 2019-02-02:
@svsdval for now try with command M301 P I D and you parameters and save with M500.

svsdval, 2019-02-03:
I found that if I set hot bed temperature to 100 *C from integrated menu (via display) - target temperature is accepted but no signal to activate mosfet (led is off)...

But if I use the M140 command target temperature is accepted and hot bed is works fine... In menu command error? O_o ?

MagoKimbra, 2019-02-05:
Now check this!

iosonopersia, 2019-02-06:
@MagoKimbra , is this now fixed also for 4_3_7 and 4_3_6 ? Can I close this issue?

svsdval, 2019-02-06:
> Now check this!
> 
> Ok.. i found the problem.. I fixed to night... Thank's!

Yup, for now it's work fine ;)

closed 2019-02-06 by iosonopersia