MKFirmware/MK4duo #647: Speed interface function is abnormal

WeiMingYu, 2018-12-24:
### Issue description

When operating the panel speed function
Whether coming in or adjusting the number of degrees
The result is an abnormal speed female fire...

### Compile errors

No error

### Firmware version
4.3.7"
- ...

Without entering the speed form...
The function is normal, the speed is normal, there is no slow motion.
but
As soon as you enter the speed form, when you move in the direction
The whole speed will become very slow

**Please be patient, we'll be trying to help you as soon as possible.**

WeiMingYu, 2018-12-25:
> ### Issue description
> When operating the panel speed function
> Regardless of operation
> Just enter the form
> Result in abnormal speed
> 
> ### Compile errors
> No error
> 
> ### Firmware version
> 4.3.7"
> 
> * ...
> 
> Just go to the adjustment speed form
> No matter what you do
> The movement of the XY axis becomes very slow movement
> 
> **Please be patient, we'll be trying to help you as soon as possible.**

WeiMingYu, 2018-12-25:
> ### 問題說明
> 操作面板速度功能時
> 無論操作
> 只需輸入表格
> 結果異常速度
> 
> ### 編譯錯誤
> 沒錯
> 
> ### 固件版本
> 4.3.7“
> 
> * ...
> 
> 只需轉到調整速度表
> 無論你做什麼
> XY軸的運動變得非常緩慢
> 
> **請耐心等待，我們會盡快幫助您。**

I checked the program
Speed-related variables in NEXTION
Is the interface that exists in pg5
Each time the action bar is pulled, pg2.vs.val=h0.val is executed.
Return value to vs variable of pg2

The vs variable is in nextionlcd.cpp, only the initial definition
With the second information
 
If (Previousfeedrate != mechanics.feedrate_percentage) {
           nexlcd.setValue(VSpeed, mechanics.feedrate_percentage);
           Previousfeedrate = mechanics.feedrate_percentage;
         }
And choose to enter gp5

   Case 5:
           Previousfeedrate = mechanics.feedrate_percentage = (int)nexlcd.getValue(VSpeed, PSTR("pg2"));
         Break;

But the speed change is only slower when executed.
AUTO HOME is normal again

In xextion.h there is another call to the original settings.
  // Manual Movement
   Constexpr float manual_feedrate_mm_m[XYZE] = MANUAL_FEEDRATE;

May I ask for answers from that direction?
Execution is like a magnification error, causing slow speed.

WeiMingYu, 2018-12-25:
> ### Issue description
> When operating the panel speed function
> Regardless of operation
> Just enter the form
> Result in abnormal speed
> 
> ### Compile errors
> No error
> 
> ### Firmware version
> 4.3.7"
> 
> * ...
> 
> Just go to the adjustment speed form
> No matter what you do
> The movement of the XY axis becomes very slow movement
> 
> **Please be patient, we'll be trying to help you as soon as possible.**
> I checked the program
> Speed-related variables in NEXTION
> Is the interface that exists in pg5
> Each time the action bar is pulled, pg2.vs.val=h0.val is executed.
> Return value to vs variable of pg2
> 
> The vs variable is in nextionlcd.cpp, only the initial definition
> With the second information
>  
> If (Previousfeedrate != mechanics.feedrate_percentage) {
>            nexlcd.setValue(VSpeed, mechanics.feedrate_percentage);
>            Previousfeedrate = mechanics.feedrate_percentage;
>          }
> And choose to enter gp5
> 
>    Case 5:
>            Previousfeedrate = mechanics.feedrate_percentage = (int)nexlcd.getValue(VSpeed, PSTR("pg2"));
>          Break;
> 
> But the speed change is only slower when executed.
> AUTO HOME is normal again
> 
> In xextion.h there is another call to the original settings.
>   // Manual Movement
>    Constexpr float manual_feedrate_mm_m[XYZE] = MANUAL_FEEDRATE;
> 
> May I ask for answers from that direction?
> Execution is like a magnification error, causing slow speed.

WeiMingYu, 2018-12-26:
I repeated the check to confirm the NEXTION code.
In the PG5 speed interface, only the return value is given to the VS variable of PG2.
And confirmed in NEXTION.CPP
Have a VS value called out
`Previousfeedrate = mechanics.feedrate_percentage = (int)nexlcd.getValue(VSpeed, PSTR("pg2"));` 

I can't find other related links when I come.

Repeat verification to confirm that the VS value /100 is not multiplied back to the basic value
Will cause the speed value to only backfill the value of 1 2 3

I should check the code of that paragraph.

WeiMingYu, 2019-01-29:
Nextion Editor in the source code
Directly
**Dims=h0.val
Pg2.vs.val=h0.val**
Throw into the VS value of pg2. There is no error here.
So it should be that the value of the call is wrong.

I guess there is something wrong here.
Because of the VSpeed value written back
The error will cause the speed value to be wrong.
Currently I am the first to disable this

**Previousfeedrate = mechanics.feedrate_percentage = (int)nexlcd.getValue(VSpeed, PSTR("pg2"));**

MagoKimbra, 2019-01-29:
Where you found?
Dims=h0.val
Pg2.vs.val=h0.val

iosonopersia, 2019-02-09:
I think this was fixed by @MagoKimbra with commit: https://github.com/MKFirmware/MK4duo/commit/f1bcd3dc4ea298cfcff352a7ab0a807e7aab872f

Please, try upgrading to the latest version and tell us if it now works correctly.

WeiMingYu, 2019-02-11:
The test results are as follows
1. Speed adjustment to 250% does accelerate
2. Speed is adjusted to 300% and the speed is slow

I changed the variable to
**Const uint16_t temp_speed = nexlcd.getValue(VSpeed, "pg2");**
Speed adjustment function is no problem to 300%
Thank you

MagoKimbra, 2019-02-11:
Thank's...

iosonopersia, 2019-02-11:
Oops! Thank you very much! You spotted a bug... A uint8_t can hold only values between 0 and 255... We should really fix this, @MagoKimbra

closed 2019-02-11 by iosonopersia