MKFirmware/MK4duo #648: Extruder manual moving (from LCD) bug

Hans007a, 2019-01-06:
Hello,

I found a bug in the last release. Anything is wrong with the manual move the extruders from the LCD display.
When I go to the move menu and select one of the three extruders many times the menu switch
back to the info display and don't show the move length selection. Many times I ca go to the selection of the move length but after select one of them the system is going to the info display.

My printer is a CoreXY with one nozzle and 3 extruders.  
Board: BOARD_RAMPS_FD_V1 with I2C EEPROM, REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
and ESP8266 on serial3

Regards

Hans

MagoKimbra, 2019-01-06:
Ok i check this!!!

Hans007a, 2019-01-06:
Sorry, I put the wrong printer data, here are the right data:

Printer is a Cartesian with one nozzle and 3 extruders.
Board: BOARD_RAMPS_FD_V1 with I2C EEPROM and REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

Thank You, that You will show this bug :)

Regards

Hans

Hans007a, 2019-03-15:
any news?
I flashed today the new release and the bug is still there...