MKFirmware/MK4duo #654: 4.3.7 won't compile when using cnc

openmy, 2019-01-15:
### Issue description
4.3.7 won't compile when enable CNCROUTER

### Compile errors
src/inc/../feature/cncrouter/sanitycheck.h:31:4: error: #error "DEPENDENCY ERROR: You have to set CNCROUTER_PIN to a valid pin if you enable CNCROUTER."

   #error "DEPENDENCY ERROR: You have to set CNCROUTER_PIN to a valid pin if you enable CNCROUTER."

### Firmware version
MK4duo's version BETA 4.3.7
Arduino IDE 1.8.7

### My setup
#define MOTHERBOARD BOARD_RAMPS_13_HFB
#define CNCROUTER
#define CNCROUTER_PIN ORIG_HEATER_BED_PIN// 

- SETTING_CHANGED_1    ORIG_HEATER_BED_PIN
- SETTING_CHANGED_2    ORIG_HEATER_0_PIN
- SETTING_CHANGED_3    "blablabla"
- ...
 
set CNCROUTER_PIN to a valid pin 
What should I set for CNCROUTER_PIN

MagoKimbra, 2019-01-15:
in configuration_pins.h
#if ENABLED(CNCROUTER)
  #define CNCROUTER_PIN NoPin
#endif
do you change this or not?

openmy, 2019-01-16:
> in configuration_pins.h
> #if ENABLED(CNCROUTER)
> #define CNCROUTER_PIN NoPin
> #endif
> do you change this or not?

change this to 
 #if ENABLED(CNCROUTER)
 #define CNCROUTER_PIN ORIG_HEATER_BED_PIN
 #endif
and delete everything in Configuration_Overall.h
compile works, thank.