MKFirmware/MK4duo #655: Compile error in version 4.3.7

ezdizzy, 2019-01-16:
### Issue description
When compiling version 4.3.7 both manually configured and using the online Configurator, the following error occurs, although previously this did not occur, 20.12.2018 compilation was successful...

### Compile errors
In file included from C:\Users\pashtet\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.23\cores\arduino/Arduino.h:28:0,

                 from c:\temp\arduino_build_757358\sketch\mk4duo.h:37,

                 from sketch\src\core\commands\commands.cpp:29:

sketch\src\core\commands\gcode/delta/g33_type2.h: In function 'void gcode_G33()':

c:\temp\arduino_build_757358\sketch\src/lcd/lcdui.h:28:52: error: initializer fails to determine size of '__c'

 #define LCD_MESSAGEPGM(x)       lcdui.set_status_P(PSTR(x))

                                                    ^

sketch\src\core\commands\gcode/delta/g33_type2.h:463:3: note: in expansion of macro 'LCD_MESSAGEPGM'

   LCD_MESSAGEPGM(deltaAC);

   ^

c:\temp\arduino_build_757358\sketch\src/lcd/lcdui.h:28:52: error: array must be initialized with a brace-enclosed initializer

 #define LCD_MESSAGEPGM(x)       lcdui.set_status_P(PSTR(x))

                                                    ^

sketch\src\core\commands\gcode/delta/g33_type2.h:463:3: note: in expansion of macro 'LCD_MESSAGEPGM'

   LCD_MESSAGEPGM(deltaAC);

   ^

exit status 1
Ошибка компиляции для платы Arduino/Genuino Mega or Mega 2560.

### Firmware version
MK4duo 4.3.7

### My setup
[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/2763086/Configuration_Overall.txt)
[Configuration_Pins.txt](https://github.com/MKFirmware/MK4duo/files/2763087/Configuration_Pins.txt)

ezdizzy, 2019-01-16:
Don't know if I did right but when I changed in the file \src\core\commands\gcode\delta\g33_type2.h
// Report settings
  PGM_P deltaAC = PSTR(MSG_DELTA_AUTO_CALIBRATE);
  SERIAL_PGM(deltaAC);
  if (verbose_level == 0) SERIAL_MSG(" (DRY-RUN)");
  SERIAL_EOL();
  LCD_MESSAGEPGM(deltaAC);

Led to such a mind
// Report settings
  PGM_P deltaAC = PSTR(MSG_DELTA_AUTO_CALIBRATE);
  SERIAL_PGM(deltaAC);
  if (verbose_level == 0) SERIAL_MSG(" (DRY-RUN)");
  SERIAL_EOL();
  LCD_MESSAGEPGM(**"deltaAC"**);
The compilation was successful, although the work was not tested, tomorrow I will try to check.

MagoKimbra, 2019-01-16:
Fixed with last commit!!!

closed 2019-01-16 by MagoKimbra