MKFirmware/MK4duo #657: Vers. 4.3.8 and Nextion 7" issue

carlonb, 2019-01-29:
### Issue description
From nextion 7" main page, pressing the "Print speed" button seems resetting all funcions. Motors, fans and all heaters goes off and need to reinit SD card. This happens if printing but even not printing. 

### Compile errors
No compile errors.

### Firmware version
MK4duo 4.3.8 25/01/2019

### My setup
Cartesian, BLtouch clone, Flux capacitor, Nextion 7" enhanced on serial 2.

NOTE: Anyway, I can say the printer prints good (if I do not touch print speed button).

carlonb, 2019-01-30:
Update:
Recompiled and reinstalled the Nextion 7" .HMI project disabling the "Print Speed" button on page 2 in order to avoid this dangerous button (at least dangerous for me).
I'm waiting for news
Thanks

Ratisse, 2019-02-01:
Same problem with Nextion 4.3" firmware, moving the speed cursor in page 5 crashes the printer board, works again after reseting it

MagoKimbra, 2019-02-01:
We did tests with the Nextion 7 "and we did not find any problem, but it was all done with Arduino DUE, so I assume that the problem is the mega that does not make it, so far no solution is possible.

carlonb, 2019-02-01:
So you think it's a processor speed problem?

iosonopersia, 2019-02-06:
Latest commit should solve this issue: https://github.com/MKFirmware/MK4duo/commit/f1bcd3dc4ea298cfcff352a7ab0a807e7aab872f

closed 2019-02-06 by iosonopersia