MKFirmware/MK4duo #658: Vers. 4.3.8 and Nextion 7" tune button issue

carlonb, 2019-01-29:
### Issue description
When printing if I go in Text menu and press Tune button I can see all item and actual values.
But if I press any of these items the values changes automatically with very low random values and ruin the printing. Here it's not possible change the values as already done for the classic LCD.

### Compile errors
No compile errors.

### Firmware version
MK4duo 4.3.8 25/01/2019

### My setup
Cartesian, Mega 2560 + Ramps 1.4, BLtouch clone, Flux capacitor, Nextion 7" enhanced on serial 2.
NOTE: Anyway, I can say the printer prints good (if I do not touch the Tune button).

carlonb, 2019-01-29:
Edit: Issue description correction: Yes, the values are changeable pressing the up/down buttons and Enter. But if I press twice the item with the actual value, this will assume a random value.

iosonopersia, 2019-02-09:
This was solved by @MagoKimbra with commit: https://github.com/MKFirmware/MK4duo/commit/b51561119f6fb1f2ee2dbc930ede058e627af2f4

closed 2019-02-09 by iosonopersia