MKFirmware/MK4duo #660: Extreme acceleration when Lin_advance is activated

Prinz12, 2019-02-01:
Hello, I have the following Corexy setup:

Board: Raads
Display: Nextion 
Number of extruders: 1
Heated bed
Firmware 4.3.8 commit 32646c2d282f26f78501d9d0dafba414b1e63688
Repetier Host V2.1.3 with 4.3.8 eeprom XML.

Only the most necessary settings were deliberately changed.

The following error occurs since the first 4.3.7 version and was not yet present in version 4.3.6. :

As soon as I activate Lin_advanced in the firmware, the movements during printing have an extremely high acceleration. Furthermore there is a faulty display of the values in the eeprom.

The values for minimum feedrate and minimum segment time are displayed twice and the values are once in red without content.

With best regards

iosonopersia, 2019-02-09:
Hi @Prinz12 !
What value of "Junction Deviation mm" did you set? Reducing the value of "Junction Deviation mm" can lower accelerations during turns o direction changes. If you want you can also lower "Max acceleration" for each axis and see if the printer starts behaving better. Should you need some advice, you can tell us your settings: default accel., max accel., junction_deviation_mm ...

As for the problem with the EEPROM values read by RepetierHost... can you please give us a screenshot that shows the problem? That would be helpful to understand what the problem really is.

Prinz12, 2019-02-09:
Hello,
Junction Deviation is deactivated and I have set the accelerations for each axis to 10 for testing purposes. Apparently the movements are executed without acceleration when the E axis is moved. Movements executed manually are executed correctly. Normally I have the acceleration at 1000 mm/s2. 

[image]

chainreaction, 2019-03-17:
### Issue description
Hmm, It seems I have the very same issue...

### Firmware version
4.3.8 rev 20022019

### My setup
* Configuration_Basic.h
`#define MECHANISM MECH_COREXY`

* Configuration_Core.h
```#define DEFAULT_MAX_ACCELERATION              {2100, 2100, 90, 8000, 8000}
#define DEFAULT_RETRACT_ACCELERATION          {3000, 3000}
#define DEFAULT_ACCELERATION          3000
#define DEFAULT_TRAVEL_ACCELERATION   3000
#define DEFAULT_XJERK 7.0
#define DEFAULT_YJERK 7.0
#define DEFAULT_ZJERK 0.4
#define DEFAULT_EJERK                   {5.0, 5.0}
```
* Configuration_Feature.h
```#define LIN_ADVANCE
#define LIN_ADVANCE_K 0
//#define JUNCTION_DEVIATION
#define JUNCTION_DEVIATION_MM 0.02
#define BEZIER_JERK_CONTROL
```

In runtime I can send M900 K0 and this would stop vibration due to unaccelerated moves. Other K-value starts vibrating mechanics again...

chainreaction, 2019-03-24:
By the way, I have same eeprom display with repetier like Prinz12 had. And it is somehow related to acceleration issue. After reflasing firmware with LIN_ADVANCE been off eeprom listing in repetier became normal.

iosonopersia, 2019-03-29:
@Prinz12 , @chainreaction 
I tried activating LIN_ADVANCE and I now have the same problem as you described (the printer shakes a lot while printing, makes really bad noise and looses steps). Setting LIN_ADVANCE factor to 0 makes the printer behave correctly. Something is surely broken in the firmware.

My setup is a pretty standard Cartesian (Prusa-like) 3D printer with Mega and Ramps.
I'll try to investigate more in my very little spare time...