MKFirmware/MK4duo #664: MKS Gen1.4 and TMC2208 Won't Compile

Burritobun, 2019-02-07:
I'm unable to compile the 4.3.7 build using a MKS gen 1.4 board, TMC2208 drivers (with programming over uart) on Platform IO. I'm getting the following errors:

`^
In file included from src\src\core\mechanics\../../../MK4duo.h:161:0,
from src\src\core\mechanics\cartesian_mechanics.cpp:29:
src\src\core\mechanics\../../../src/feature/tmc/tmc.h: In static member function 'static void TMC_Stepper::set_stealth_amplitude(MKTMC*, uint8_t)':
src\src\core\mechanics\../../../src/feature/tmc/tmc.h:471:11: error: 'class MKTMC' has no member named 'pwm_ampl'
st->pwm_ampl(stealth_amplitude_val);
^

*** [.pioenvs\megaatmega2560\src\src\core\heater\heater.cpp.o] Error 1
*** [.pioenvs\megaatmega2560\src\src\core\fan\fan.cpp.o] Error 1
*** [.pioenvs\megaatmega2560\src\src\core\eeprom\eeprom.cpp.o] Error 1
*** [.pioenvs\megaatmega2560\src\src\core\endstop\endstops.cpp.o] Error 1
In file included from src\src\core\commands\gcode/gcode.h:124:0,
from src\src\core\commands\commands.cpp:30:
src\src\core\commands\gcode/feature/m940_m942.h: In function 'void gcode_M941()':
src\src\core\commands\gcode/feature/m940_m942.h:114:17: error: 'class MKTMC' has no member named 'chm'
stepperX->chm(parser.value_bool());
^
src\src\core\commands\gcode/feature/m940_m942.h:122:17: error: 'class MKTMC' has no member named 'chm'
stepperY->chm(parser.value_bool());
^
src\src\core\commands\gcode/feature/m940_m942.h:141:18: error: 'class MKTMC' has no member named 'chm'
stepperE0->chm(parser.value_bool());
^
In file included from src\src\core\commands\../../../MK4duo.h:61:0,
from src\src\core\commands\commands.cpp:29:
src\src\core\commands\../../../src/lib/macros.h: In instantiation of 'constexpr void NOLESS(A&, B) [with A = unsigned int; B = int]':
src\src\core\commands\gcode/probe/m851.h:51:36:   required from here
src\src\core\commands\../../../src/lib/macros.h:276:94: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
template <class A, class B> static inline constexpr void NOLESS(A& a, const B b) { if (a < b) a = b; }
^`

Burritobun, 2019-02-07:
Here's my project file [https://www.dropbox.com/sh/hu1etp3qk7zwffc/AAAACpxUIB6lprfz05C9lYzZa?dl=0](url)

MagoKimbra, 2019-02-07:
Fixed!

closed 2019-02-07 by MagoKimbra