MKFirmware/MK4duo #665: Delta. Error in commands G0, G1 M666, M851 when they are sent from the program Repetier-Host

vsvservice, 2019-02-08:
### Issue description
Delta. Error in commands G0, G1 M666, M851 when they are sent from the program Repetier-Host v2.1.3

When the g-code : G0/G1, M666, M851 commands are sent from the Repetier-Host program, the printer does not accept the numeric parameter values correctly. 
For example: sent G0 X10 y20 z130, printer accepts x0 y0 z0. (badly) 
For example: sent M666 H298.23, the printer adopts H0.0 (bad)
For example: sent M851 x-2 y-3 z - 12, printer accepts x0 y0 z0 (bad)
This error work settings:
M666 X Y Z a B C I J K U V W R d O P H-printer always accepts 0
M851 x Y Z-always printer accepts 0
M666 s L-takes correctly.
For example: sent m666 s160 L90 printer accepts s160 L90 (good)
M851 s F R-accepts correctly.
For example: sent M851 S400 F3000 R2 printer accepts S400 F3000 R2 (good)

This problem exists in version 4.3.7 and 4.3.8 version.
In version 4.3.6 this problem is not present, everything works correctly.
### Compile errors
No compile errors.

### Firmware version
MK4duo - Firmware configurator tool (Rev 4.3.5) - Firmware Version 4.3.8 (Rev 26012019)

### My setup
Delta, DUE + RuRamps4Dv13, LCD2004 or Full Grafic 12864

MagoKimbra, 2019-02-08:
I have the delta as a printer, I use repetier host 2.1.3 and absolutely nothing of what you said ...
If I give M666 H210.1 goes, the G1 X Y Z commands go, surely there is some wrong setting somewhere.

vsvservice, 2019-02-08:
What to prove that I have happens as I described? After changing the firmware version, I always make m502 commands. M500.

vsvservice, 2019-02-08:
I also tried another program Pronterface 2014.08.01. Everything happens as I described. Configuration of the new version was performed in the Configurator. Then I changed some options in the manual. All actions and settings are similar to version 4.3.6.

vsvservice, 2019-02-08:
I found a solution to the problem. 
I have the configuration option enabled:
 #define INC_MODE_SUPPORT. 
 If this option is commented out then the problem disappears. 
If you send the G21 command with this option enabled, the problem disappears.

MagoKimbra, 2019-02-08:
Ok

closed 2019-02-08 by iosonopersia