MKFirmware/MK4duo #681: MK4duo - Firmware configurator tool (Rev 4.3.5) - Firmware Version 4.3.8 (Rev 20022019)

hagre, 2019-03-09:
### Issue description
Using the Firmware configurator tool (Rev 4.3.5) - Firmware Version 4.3.8 (Rev 20022019) for my Anycubic Kossel Linear Plus compiling/working errors happend.

### Compile errors
I have to manualy activate:
#define LCD_BED_LEVELING
#define MESH_EDIT_MENU
and 
#define G26_MESH_VALIDATION // only when using UBL

### Firmware version
MK4duo's versions that have this issue.

### My setup
Anycubic Kossel Linear Plus with
#define REPRAP_DISCOUNT_SMART_CONTROLLER
and
#define Z_PROBE_FIX_MOUNTED on Zmin pin (orginal probe ver2)

iosonopersia, 2019-03-10:
Please, give us more info: what compiling errors are you exactly getting?

hagre, 2019-03-10:
I have tried now many times to reproduce the compiling errors i had some days ago, but (you can be happy) i had no compiling problem any more. So the Problem is solved. Thank you for that Software!

I just wanted to add: the preconfigured setting you are offering in the online configuration tool for "Anycubic Linear Plus" are looking for me to belong to the build volume of the "pulley" or small "linear" version. Some small adjustments regarding the extruder direction, ... are needed (Depends on the exact version of the Printer and the z probe). If someone is interrestet on my final settings i add the file.

best regards

closed 2019-03-10 by hagre

MagoKimbra, 2019-03-11:
Yes add the file thank's

hagre, 2019-03-12:
Anycubic Kossel Linear Plus - 
After configurator i have manualy activate additionaly:
#define LCD_BED_LEVELING // in the first part of the file AND  "lcdbedlevel": "1" in the configurator part
#define MESH_EDIT_MENU //ONLY in the first part of the file, i don't know what to change in the configurator part.

hagre, 2019-03-12:
[Configuration_Overall_upload.txt](https://github.com/MKFirmware/MK4duo/files/2955898/Configuration_Overall_upload.txt)