MKFirmware/MK4duo #682: Compile Errors with USB_FLASH_DRIVE_SUPPORT

erdgncakmak, 2019-03-09:
### Issue description
I'm having compile errors when choose USB_FLASH_DRIVE_SUPPORT instead of sdcard with BOARD_RAMPS_13_HFB and BOARD_RAMPS_SMART_HFB boards.

### Compile errors
[BOARD_RAMPS_SMART_HFB Compile Errors.txt](https://github.com/MKFirmware/MK4duo/files/2948349/BOARD_RAMPS_SMART_HFB.Compile.Errors.txt)
[BOARD_RAMPS_13_HFB Compile Errors.txt](https://github.com/MKFirmware/MK4duo/files/2948350/BOARD_RAMPS_13_HFB.Compile.Errors.txt)

### Firmware version
Version 4.3.8 revision 20022019

### My setup
Changed anything in configurations except board type and USB_FLASH_DRIVE_SUPPORT

iosonopersia, 2019-03-24:
Hi @erdgncakmak , do you still have this error, even after all the changes that have been made lately? Please download the latest version and retry.

erdgncakmak, 2019-03-25:
Hi @iosonopersia . I tried and did not receive an error. Thanks.

closed 2019-03-25 by iosonopersia