MKFirmware/MK4duo #685: Fan connected to D4 pin works only on full speed

chainreaction, 2019-03-10:
### Issue description
Fan connected to D4 pin works only on full speed.
Fan is connected with the help of simple current extender on power bipolar transistor.
M42 P4 S255 turns fan on full speed. Any other S values turns it completely off.
Electrical setup was tested on Marlin 1.0.0-RC8 and works as expected. Any values of S do it's job well.

### Firmware version
4.3.8 rev 20022019

### My setup
The board is MKS GEN 1.4

* Configuration_Feature.h
`#define SOFT_PWM_SPEED 0`
`// #define ENABLE_SERVOS`

* Configuration_Basic.h
`#define MOTHERBOARD BOARD_RAMPS_13_HFB`

Thank you!

MagoKimbra, 2019-03-10:
But let's manage it from the firmware and not with the M42 command that uses the digital write on off to test the pins ...

chainreaction, 2019-03-10:
Do u mean reassign D4 to ORIG_FAN?_PIN pin ans use M106 instead of M42?

MagoKimbra, 2019-03-10:
No ORIG_FAN, in configuration_pins you can assign all fan pins
#define FAN0_PIN            ORIG_FAN0_PIN
#define FAN1_PIN            ORIG_FAN1_PIN
#define FAN2_PIN            ORIG_FAN2_PIN
#define FAN3_PIN            ORIG_FAN3_PIN
#define FAN4_PIN            ORIG_FAN4_PIN
#define FAN5_PIN            ORIG_FAN5_PIN
Put 4 in FAN0_PIN or what you want...

chainreaction, 2019-03-10:
* Configuration_Pins.h
`#define FAN1_PIN             4`
* Configuration_Feature.h
`#define FAN_MIN_PWM 60`
`#define FAN_MAX_PWM 255`
`#define FAN_PWM_FREQUENCY 250`
`#define FAN_KICKSTART_TIME 200`

Some strange result. Fan is working for about of a half second with M106 and then stops. Even with full speed M106 P1 S255.

MagoKimbra, 2019-03-10:
Because the fan1 is used for default in auto fan for cooler Hotend.
in configuration_feature
// AUTO FAN - Fans for cooling Hotend or Controller Fan
// Put number Hotend in fan to automatically turn on/off when the associated
// hotend temperature is above/below HOTEND AUTO FAN TEMPERATURE.
// Or put 7 for controller fan
// -1 disables auto mode.
// Default fan 1 is auto fan for Hotend 0
#define AUTO_FAN { -1, 0, -1, -1, -1, -1 }

chainreaction, 2019-03-10:
Defining 
`#define AUTO_FAN { -1, -1, -1, -1, -1, -1 }`
solved the issue!

Thank you very much!

closed 2019-03-10 by chainreaction