MKFirmware/MK4duo #686: The problem of movement on the axis Z

pirat777knj, 2019-03-11:
Hi.
There was a problem of moving along the zox axis.
After the G28 command is applied, the table is lowered, and then the work is broken.
The servo does not extend, the table no longer moves at all. Only restart helps.
Worked with this configuration on 4.3.5
Collected through the configurator, then checked manually. The limit switches are fine, setting them up too.

Djac1974, 2019-03-11:
Hello,
Actually I also noticed this problem on a CORE XY (Hypercube Evolution),
the initialization begins, the stopper picks up the switch, the table  stopped , you have to go down again and start again.
the problem also existed on version 4.3.7
thanks in advance.

pirat777knj, 2019-03-11:
Yes. On 4.3.7 there is also a problem. Only I have Cartesian) put 4.3.5 - all the rules, but the laser does not work well (

MagoKimbra, 2019-03-11:
Sorry please explain the problem.. I have cartesian and more people with core and haven't problem...
Remember this software save in eeprom more valor, check if this valor is exactly. And check endstop with M119...

pirat777knj, 2019-03-11:
The endstops are normal. The M119 command returns "Not Triggered" in a free condition. It works as it is. But on versions 4.3.7 and above the G28 and G29 commands work not.
It looks like this:
Home X - Home Y, move to X100 Y100, move Z10 and end. More printer is not moving. At all. It helps only reboot button or power off.

pirat777knj, 2019-03-11:
Forgot to add. After that, the printer even stops responding to commands. But it really hangs (

MagoKimbra, 2019-03-11:
Ok Board?
Use host or only lcd??

pirat777knj, 2019-03-11:
I have radds 1.5. And use only host.

MagoKimbra, 2019-03-11:
Ok. When does the home do not report that the home has failed?

pirat777knj, 2019-03-11:
Communication with the printer is lost immediately after the table rises along the Z axis. There is one message in the port monitor: timeout
Even the servo does not move (
Tomorrow in the evening I can take a video of how it looks.

Djac1974, 2019-03-12:
Hello,

 

I use Radds1.5 with arduinno due et nextion 7’.

I encounter the problem often with nextion 7 'and sd card.

 

Best regards.

Djac1974, 2019-03-12:
Hello,

 

The table moves, it reaches the sensor and never information from the host as if we had lost communication.

I work with a RADDS1.5, on a chassis hypercube evolution, and nextion 7 '.

 

Best regards.

Djac1974, 2019-03-12:
Hello,

 

correct

I have the same problem

 

Best regards.

Djac1974, 2019-06-01:
I go back to this topic, when I make a homing request on the nextion screen, the bed touches the Zmin sensor, and hangs.
I use a RADD1.5, independent Z min and Bltouch.
At the beginning of printing during the initialization I never have the problem.