MKFirmware/MK4duo #687: Version 4.3.9 beta Compile errors

Djac1974, 2019-03-11:
### Issue description
I change the beta version, add the new definition in the overhall, now I have a compile default.

### Compile errors
Arduino : 1.8.5 (Windows 7), Carte : "Arduino Due (Programming Port)"

ATTENTION : la bibliothèque TMCStepper prétend être exécutable sur la (ou les) architecture(s) (avr) et peut être incompatible avec votre carte actuelle qui s'exécute sur (sam).
In file included from sketch\src\core\endstop\../../../src/platform/platform.h:33:0,

                 from sketch\src\core\endstop\../../../MK4duo.h:112,

                 from sketch\src\core\endstop\endstops.cpp:27:

sketch\src\core\endstop\endstops.cpp: In static member function 'static void Endstops::update_software_endstops(AxisEnum)':

sketch\src\core\endstop\endstops.cpp:649:66: error: 'struct mechanics_data_t' has no member named 'home_offset'

         SERIAL_MV(" axis:\n data.home_offset = ", mechanics.data.home_offset[axis]);

                                                                  ^

sketch\src\core\endstop\../../../src/platform/common/communication.h:118:56: note: in definition of macro 'SERIAL_VAL'

 #define SERIAL_VAL(val, ...)                Com::print(val, ## __VA_ARGS__)

                                                        ^

sketch\src\core\endstop\endstops.cpp:649:9: note: in expansion of macro 'SERIAL_MV'

         SERIAL_MV(" axis:\n data.home_offset = ", mechanics.data.home_offset[axis]);

         ^

sketch\src\core\endstop\endstops.cpp:650:53: error: 'class Core_Mechanics' has no member named 'position_shift'

         SERIAL_MV("\n position_shift = ", mechanics.position_shift[axis]);

                                                     ^

sketch\src\core\endstop\../../../src/platform/common/communication.h:118:56: note: in definition of macro 'SERIAL_VAL'

 #define SERIAL_VAL(val, ...)                Com::print(val, ## __VA_ARGS__)

                                                        ^

sketch\src\core\endstop\endstops.cpp:650:9: note: in expansion of macro 'SERIAL_MV'

         SERIAL_MV("\n position_shift = ", mechanics.position_shift[axis]);

         ^

exit status 1
Erreur de compilation pour la carte Arduino Due (Programming Port)

Ce rapport pourrait être plus détaillé avec
l'option "Afficher les résultats détaillés de la compilation"
activée dans Fichier -> Préférences.

### Firmware version
Version 4.3.9 beta

### My setup

[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/2951744/Configuration_Overall.txt)
[Configuration_Pins.txt](https://github.com/MKFirmware/MK4duo/files/2951745/Configuration_Pins.txt)

Djac1974, 2019-03-11:
Thank you for this change
but continuing compilation other problem

### **Compile errors**

Arduino : 1.8.5 (Windows 7), Carte : "Arduino Due (Programming Port)"

ATTENTION : la bibliothèque TMCStepper prétend être exécutable sur la (ou les) architecture(s) (avr) et peut être incompatible avec votre carte actuelle qui s'exécute sur (sam).
In file included from sketch\src\core\printer\../../../src/inc/pins.h:26:0,

                 from sketch\src\core\printer\../../../MK4duo.h:105,

                 from sketch\src\core\printer\printer.cpp:29:

sketch\src\core\printer\printer.cpp: In static member function 'static void Printer::setup()':

sketch\src\core\printer\../../../src/inc/../boards/1401.h:253:13: error: 'RADDS_EXT_VDD_PIN' was not declared in this scope

   OUT_WRITE(RADDS_EXT_VDD_PIN, HIGH); \

             ^

sketch\src\core\printer\printer.cpp:125:5: note: in expansion of macro 'MB_SETUP'

     MB_SETUP;

     ^

sketch\src\core\printer\../../../src/inc/../boards/1401.h:254:13: error: 'RADDS_EXT_VDD2_PIN' was not declared in this scope

   OUT_WRITE(RADDS_EXT_VDD2_PIN, HIGH);

             ^

sketch\src\core\printer\printer.cpp:125:5: note: in expansion of macro 'MB_SETUP'

     MB_SETUP;

     ^

exit status 1
Erreur de compilation pour la carte Arduino Due (Programming Port)

Ce rapport pourrait être plus détaillé avec
l'option "Afficher les résultats détaillés de la compilation"
activée dans Fichier -> Préférences.

iosonopersia, 2019-03-24:
These errors should not be a problem anymore. If you get other error messages, tell us here.
As for now, I'm closing this issue.

commit 6472f649 (2019-03-11): Fix #687 and clear code
commit 80bc4ad6 (2019-03-11): Fix #687
closed 2019-03-24 by iosonopersia