MKFirmware/MK4duo #689: Bed and chamber reaseting their target temp to lower ones

chainreaction, 2019-03-16:
### Issue description
Short time after I have set bed and chamber target temp they been reseted to lower ones.
I'm using 
`M140 S110`
and
`M191 S50`
After that bed -> 100 and chamber -> 45

### Firmware version
4.3.8 rev 20022019

### My setup
* Configuration_Temperature.h
```
#define HEATER_0_MAXTEMP 250 // (degC)
#define HEATER_1_MAXTEMP 250 // (degC)
#define HEATER_2_MAXTEMP 250 // (degC)
#define HEATER_3_MAXTEMP 250 // (degC)
#define BED_MAXTEMP      115 // (degC)
#define CHAMBER_MAXTEMP  60 // (degC)
#define COOLER_MAXTEMP   35  // (degC) 
```

MagoKimbra, 2019-03-17:
The values of max temp are values beyond which it must not go otherwise it goes into error. Naturally for obvious reasons those values cannot be set as target, because even a single tenth of degree would be enough that the firmware goes into error. So there is a 15 degree margin for settable temps. If max temp is 115 the maximum temperature that can be set is 100. In the dev version the margin has been reduced to 10 °.

chainreaction, 2019-03-18:
Ok, thx for the comment!

closed 2019-03-18 by chainreaction