MKFirmware/MK4duo #696: Part Cooling Pin

Buddycarpenter, 2019-04-05:
I'm running firmware version 4.3.8 on my custom built printer.  The mother board is an Arduino Due.  And a modified ramps 1.4.  I'm trying to setup a parts cooling fan that will work with the Simplify3D slicer.  Under features, I found where to setup the hotend fan (cold end) for each hotend or for the controller, but have not found a pin for the part cooling fan. 

Can someone help me out with setting up the part cooling fan?  Simplify3d has a tab for controlling the fan with the slicer software.

### Firmware version
4.3.8

### My setup

[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/3049760/Configuration_Overall.txt)
[Configuration_Pins.txt](https://github.com/MKFirmware/MK4duo/files/3049761/Configuration_Pins.txt)

MagoKimbra, 2019-04-05:
The fan0 is normal used for cooling object pla. Now setting the fan0 not auto fan on hotend and use fan1 fan 5 for auto fan..

Buddycarpenter, 2019-04-06:
Thank you very much.  Worked great.

William

closed 2019-04-06 by Buddycarpenter