MKFirmware/MK4duo #697: BIGTREETECH SKR V1.3 (1.1) LPC1768 when support will realised?

ZhenyaKa, 2019-04-12:
Marlin 2.0 have support of this board.

When Mk4duo will support it?

MagoKimbra, 2019-04-20:
No sorry, LPC1768 not arduino compatible...

closed 2019-04-20 by MagoKimbra

ZhenyaKa, 2019-04-20:
How Marlin 2.0 can support it? It is arduino based too.
Strange answer.

ZhenyaKa, 2019-04-21:
http://exploreembedded.com/wiki/Arduino_Setup_for_Explore_M3

MagoKimbra, 2019-04-23:
Those libraries are not for the LPC1768 processor, but for a card that mounts that processor, but uses only a part.
In fact Marlin 2.0 for the LPC1768 processor uses platformio with other libraries for the processor.
Unfortunately, to do this step I need to transfer the whole project to eclipse or platform I abandon Arduino IDE and also have cards to try, I don't have the time or the desire to do it. On Marlin there are many people who contribute everyone has his card to make tests and can modify and adjust knowing that things go, I alone cannot. For this reason I stop at the cards on the arduino DUE that I have and I know well as a processor, the rest I can't ...