MKFirmware/MK4duo #701: The time when printing is displayed with errors and printing errors.

pengyong2000, 2019-04-21:
Hello, I have the following CARTESIAN setup for my CR-10(Creality):

Board: Ramps1.4
Display: RepRapDiscount Full Graphic Smart Controller
Number of extruders: 1
Heated bed
Firmware 4.3.7
Set in the software to BOARD_RAMPS_13_HFB and REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER.

Only the most necessary settings were deliberately changed.

Machine homing and heating is normal.

The file is printed and the runtime display is wrong.

If the printed file is not large, it can be printed normally, and the total time for printing the file is correct.

If the printed file is large, the printing will be wrong in the middle of the printing, and the extrusion machine will pull back and the motor will move.

If the restart function is used, even if the printed file is not large, although it can be printed, the total time of printing will be several years.

My English is not very good, I hope I expressed the correct meaning.

With best regards

iosonopersia, 2019-04-28:
Hi @pengyong2000 , I don't understand this:

> If the printed file is large, the printing will be wrong in the middle of the printing, and the extrusion machine will pull back and the motor will move.

What happens exactly? Are you printing from USB or from SD card?

pengyong2000, 2019-04-28:
Hi @iosonopersia，thank you for taking the time to answer my question.

I found the reason because CR-10 (Creality)'s slicing software caused it to add unnecessary commands.
I am working fine with other slicing software. I haven't found a line yet because the file is too big.

The time problem is because I opened the restart function, and this line of code:
 job_info.print_job_counter_elapsed = print_job_counter.duration() * 1000UL;

Will cause each power outage to be multiplied by 1000 times, which should be:
job_info.print_job_counter_elapsed = print_job_counter.duration();

The latest version of MK4duo-4_3_9's restart function can only be powered off once like Marlin, and there will be problems with rising and then printing. Moreover, the SD card cannot be unplugged when it is powered on, and cannot be initialized normally unless it is turned on again.

Thank you for your attention to my question.

MagoKimbra, 2019-04-28:
thank's for bug duration. I do not understand the problem with SD.
In my printers i remove sd and reinsert and have not problem.

pengyong2000, 2019-04-28:
I did not define the detection pin of SD. I can't initialize it when I pull it out and reinsert it. I have to shut it down and then turn it on to detect SD. Initialization does not work. I tried to read the latest updated code, but I didn't find the reason.

MagoKimbra, 2019-04-28:
But have you enable #define SD_DISABLED_DETECT?

pengyong2000, 2019-04-28:
I tried to modify them:
#define SD_HALF_SPEED
#define SD_CHECK_AND_RETRY
#define SD_DISABLED_DETECT
But the situation is the same. Maybe I have neglected something, find time and try again.

carlonb, 2019-04-28:
Appear to be the same issue here #702

pengyong2000, 2019-04-28:
Okay thank you. Let me try it.

MagoKimbra, 2019-04-28:
Ok you both have the MEGA as a processor, check with my MEGA if it has the same problem.

pengyong2000, 2019-04-28:
Ok, thank you for providing such a good software, there will be many questions to ask you in the future.

closed 2019-05-26 by MagoKimbra