MKFirmware/MK4duo #702: SD Card Reader NOT working

Hans007a, 2019-04-23:
Hello,

first, thank You for the solving the other bugs :)

But after update to the newer versions my SDCard slot stopped to work...
I flashed my early backuped version (#define FIRMWARE_REVISION         "10022019") to test my board and it works ok :)
I know, You changed anything in the SDCard source code, maybe at the pins, please check it.

I attached my settings for my cartesian printer in my other issue, but if You ned I can give You my
full source code backup.

Regards

Hans

carlonb, 2019-04-23:
+1
MK4Duo vers. 4.3.9, Cartesian with Nextion 7"
My SD card is correctly read at printer start-up, but if I remove the SD and reinsert it again, the card it's no more recognized.
With vers. 4.3.8 SD card is correctly read.
NOTE: If I select a file to be printed, the Nextion 7" will display a strange char followed by a "?", what it means?

Hans007a, 2019-04-23:
I found a setting in Configuration_Feature.h:
`#define SD_HALF_SPEED`
and for me this issue can be closed :)

regards

Hans

carlonb, 2019-04-23:
I already tried with full speed, half speed and eighth speed but nothing solve the issue

carlonb, 2019-04-24:
Further confirmations for Alberto Mago:
RAMPS Mega 2560
Nextion 7 "
#define SD_detect_inverted (my contact is closed to ground when the SD is out and open when it is inserted)

With the vers. 4.3.8 the SD is read correctly, and if I change SD without turning off the printer and pressing (randomly because I have not yet understood the sequence) "Change SD", "init SD" the new SD just inserted is correctly read.
All this does not happen with the vers. 4.3.9 of 04/22/2019, I repeat that I have to switch off or reset to make it recognized.
I also tried a quarter and a sixteenth of the speed but it doesn't change anything.
I tried not to reverse SD_detect but it doesn't change anything.
The problem could only concern those who use the Nextion ...
I attach config_overall.

I quote the previous note so as not to lose it: When I select a print from SD, on the Nextion 7 "a strange character appears and a"? ", What is it?

Thanks bye

MagoKimbra, 2019-04-29:
Ok tried both on a ramps with lcd graph, and on wanhao d6. I remove the SD card and then put it back and read it immediately.
What other test can I do to simulate your problem?

carlonb, 2019-04-29:
Thanks Alberto,
Have you tested with my config_overall as above?
The strange think is that with 4.3.8 all is good.

MagoKimbra, 2019-04-29:
Yes, but 4.3.8 have old sfdat, in 4.3.9 i put new sdfat library...
Test new command for format sd card directly from firmware.

carlonb, 2019-04-29:
Uncommented #define ADVANCED_SD_COMMAND,  installed 4.3.9
PID done
How format the SD with M39 command?

carlonb, 2019-04-29:
Is it mandatory update the Nextion fw for testing the SD? I've not already updated my Nextion.
Anyway, the SD reader do not work, If I insert a Card after swith the printer On, the card is recognized, but If I extract this SD and I insert a second SD I see the file list of first card. Then If I launch a print on the false file shown, the system crashes.
The SD refresh command on display do nothing...

MagoKimbra, 2019-04-29:
A ok you insert another SD card, this is important.. Now test this...

carlonb, 2019-04-29:
Yes, but also with the same card I need to switch the printer On and Off…..

carlonb, 2019-04-29:
Update:
Uploaded the new Nextion s/w but the SD issue not solved.

MagoKimbra, 2019-04-29:
The nextion has nothing to do with the SD ...
On my card with the nextion if I remove the sd and put it back it works perfectly, but I haven't tried to change SD ...

carlonb, 2019-04-30:
Alberto,
The main problem is that the SD card is not read if I remove it and then reinserted. I need to reboot the printer to read again this SD.
The second issue (but I think it's the the same problem) if I remove one SD and insert a second SD it's not initialized. On the display I see the file list of the first SD card.
Where are the routine to init/read the SD in firmware? I will try to investigate this issue….
Bye

MagoKimbra, 2019-04-30:
The routine is the same with any processor and SD. When the SD is removed it is disassembled and reinserted when it is reinserted, and everything works via the pin.
Try disassembling the SD with the M22 command and then reassemble it with M21 and check if it works..

carlonb, 2019-04-30:
With M22 (release card) and then M21 (Init card) using the Keyboard button on Nextion all is reinitialized and I can read the 2 card, all OK.
The same think sending M22 then M21 commands with Repetier Host. all OK.
If I use the Text Menu of Nextion, selecting Change SD Card, I see on Repetier "echo:enqueueing M21 and nothing happen. I think that Change SD card must send before M22 and then M21.

NOTE: As I use the SD detect switch, when I insert or remove the card on Repetier log window I do not see nothing… is it correct?

MagoKimbra, 2019-04-30:
Which Nextion card change menu do you mean?
If you remove the sd in the host log you don't read anything, but on the display you read SD remove.

carlonb, 2019-04-30:
On text menu of Nextion… second line there is "Change SD card", pressing it, I see on host "enqueueing M21"

MagoKimbra, 2019-04-30:
Yes you are right, in that case he must give the command M22 not M21.
If yuo open menu_main.cpp in src/lcd/menu change the line
MENU_ITEM(gcode, MSG_CHANGE_SDCARD, PSTR("M21"));  // SD-card changed by user
and put M22 not M21.
Try... Thank's

MagoKimbra, 2019-04-30:
But you don't have the pin detect then?

carlonb, 2019-04-30:
Disabled SD_Detect pin! (it was enabled but not seen by s/w).
Changed M21 with M22 in function MENU_ITEM(……) and now  SD change by user it's ok!

My issue now is that I HAVE the SD detect pin, it's inverted (open when SD inserted), but for Nextion users is not defined, my SD detect pin is on pin D49, I've tried to define it, but it's not working.
Please tell me how to test the input pins I do not remember how to do it :)

carlonb, 2019-04-30:
Resolved issue !!! :)
I had 3 problems in one time:
1) My SD detect switch is =0 when SD inserted =1 when SD removed so it's NOT inverted (I before said the reversed).
2) The SD detect switch on my card reader some time do not close the contact.
3) With Nextion display SD detect pin is not defined (I added the define in configuration_pins_h).
So:
If I do not use the SD detect switch (#define SD_DISABLED_DETECT) I can mount/dismount the SD card by the Nextion menu'.
If I use the SD detect feature, automatically the SD is detected when inserted.

Alberto please add the SD_detect_pin if Nextion is selected.

MagoKimbra, 2019-04-30:
The problem is that the nextion doesn't have the sd, but it's an external sd and I don't know what the pin is, how can I define it?

carlonb, 2019-05-01:
Alberto you are right,
May be you can define a NoPin if an user select the Nextion display, then you can add a sanity check to tell the user to define a free pin (if available) as a reminder…. But I don't know if this is possible...

MagoKimbra, 2019-05-01:
Into the file your board for example ramps HFB 33.h you found:
#define SD_DETECT_PIN              NoPin
Now change this with your pin for SD_DETECT_PIN.

carlonb, 2019-05-01:
Yes, but now I've defined it in configurations_pins and I think it's the same.

MagoKimbra, 2019-05-01:
Yes it's same.