MKFirmware/MK4duo #707: Any possibility of MK4duo adding LPC1768 ?

doggyfan, 2019-04-28:
Just been having a look at MK4duo's source code for the first time.

It's sooo clean compared to Marlin v2, really nice !   I'm about to try it on my MKS Gen 1.4 board, 

I also have a BigTreeTech SKR 1.3 board (LPC1768 100MHz MPU) which would be lovely to run MK4duo on. I don't suppose you have any plans for adding support for the LPC1768 boards ?

Anyway, I've been hard/soft/firm designer for a few decades now, and love nice clean code such as MK4duo. You may well have converted myself from Marlin to MK4duo :)

Well done guys and gals !

MagoKimbra, 2019-04-28:
The problem is this, there are neither boys nor girls, only me. And I can't start writing again for another processor and, in addition, having to change also the type of compilation by going to platform, I don't have the time or the means, I should take cards with that processor to do tests. So I stopped at the SAM3E processor (Arduino DUE)..

doggyfan, 2019-04-28:
That's OK @MagoKimbra, I didn't realise MK4duo is done by only yourself. You've done a wonderful job.

I totally understand, adding more CPU's would be a big task for one person. I've been there myself I know what it's like.

closed 2019-04-28 by doggyfan