MKFirmware/MK4duo #708: [REQUEST] Granular Controls for Delta Calibration / Levelling

lohiaprateek, 2019-04-28:
Hey, Thank you for this wonderful firmware, IMHO its much better than Marlin due to the ease of use of configuration and smoother prints in comparison to Marlin.

I am using a custom made Delta printer which uses a removal probe with 0,0 offset on X and Y so as to have 0 tilt while calibrating. But whenever I use either calibration 1 or 2 method, it fails to calibrate and asks me to check radius. My printable radius is 85 and I usually 80% of printable radius for calibration, but the full printable radius for bed leveling. I believe, Marlin has a different radius for Calibration and for Leveling. So I usually manually change probe radius before calibration and leveling. 

Also if using the dc42 algorithm for calibration if there could be an input to choose default number of factors and number of points.

It would be nice if calibrations and leveling could be independent of a computer, but I need to make the changes to radius and factors and points every time I calibrate and level. 

I could always use the Custom user entry to make those changes, but it adds complication to a very nice and orderly firmware. 

Thanks

MagoKimbra, 2019-05-26:
MK4duo have probeable radius.. See M666 for change it!

lohiaprateek, 2019-05-29:
Let me explain it a little differently. After I flash the firmware in which Delta print radius and probe radius is same, I want to run a G33 from the display. Both calibration methods dont run. 

On connecting printer to computer and initiating a G33 Method 1, it probes 3 points and then stops, in method 2 it throws an error as inplausible radius. I thus have to reduce the probe radius by using the M666 command to reduce it by .1 mm to initiating a probe. This successfully runs both strategies. 

Also in Calibration strategy 1, as recomended by dc42, 6 factor should be more than enough for calibration as the 7th factor (rod length) can be measure accurately on a jig and it need not change, but running a G33 from the display will by default use 7 factors, thus needing a computer to calibrate. 

If we could have both Probe radius and number of factors for calibration method 1 in the configuration. calibration could be completed from the display.

MagoKimbra, 2019-05-29:
The probe radius is saved in eeprom so there is no need to change it every time.
As for G33, the command provides the F sub-command for the number of factors, just give G33 F6.

lohiaprateek, 2019-05-29:
The first time change has to be done by a computer, and running G33 F6 P10 cannot be done from a display, there is a need for a computer. 

Hence, my request was if both could somehow be incorporated in the firmware such that there is no need of a computer. One could just use the display to calibrate a printer and then begin printing.

MagoKimbra, 2019-05-29:
My vision of the firmware is very different from the others. For me the calibrations and the various settings should be done only by the computer to get all the information via log. Once done then you can use the display to just print.
As for the command you can create custom menus where you define the commands you want:
// # define CUSTOM_USER_MENUS

#define USER_SCRIPT_DONE "M117 User Script Done"

for example:

#define USER_DESC_1 "Calibrate Delta"
#define USER_GCODE_1 "G28 ng33 F6 P10"

lohiaprateek, 2019-05-29:
Exactly what I had planned to do, by creating a user script, but from there begins duplication for an average end user, as it will show a Delta Calibration section on the display with details, and also in the user script. 
I agree with you entirely that calibration should be initiated on the computer as logs can be seen properly on the host, then could be blank out the delta calibration section on the display ?

Sorry for being a bit stubborn. 

The method I was using till now was to change on boot code to M666 P54.9 (55 is the print radius on the printer) and changed the MENU_ITEM(gcode, MSG_DELTA_AUTO_CALIBRATE, PSTR("G33")); to run a G33 F6 P10 in file menu_delta_calibrate. This in an ad-hoc method of reducing duplication and also not needing a computer to start prints.

MagoKimbra, 2019-05-29:
But I'm sorry, but do you calibrate every time?
Why don't you save the values in eeprom and then don't do it anymore.
I have the delta and I do the calibration once a month or maybe more !!!

lohiaprateek, 2019-05-29:
I frequently change parts on my printers as I am testing and completing my self designed delta. 

Also out of curiosity, what calibration method are you using the calibrate your delta. I get varying results from either methods.

MagoKimbra, 2019-05-29:
That of DC42, the one taken by Marlin does not convince me and I also get strange results.

MagoKimbra, 2019-05-29:
I add this:
#define DELTA_AUTO_CALIBRATION_1_DEFAULT_FACTOR 6
#define DELTA_AUTO_CALIBRATION_1_DEFAULT_POINTS 7

closed 2019-05-29 by MagoKimbra

lohiaprateek, 2019-05-29:
Thank you. This should be perfect.