MKFirmware/MK4duo #709: Compile error, or quite possibly user error (new to the code) - TMC2208 hybrid_thrs error

doggyfan, 2019-04-29:
I'm using TMCstepper v0.3.2

I've only listed the changes to the config files I've done.

--------------------------------------------------------------
Compile error I get ...

In file included from sketch\MK4duo.h:166:0,

                 from C:\Projects\3D Printer\MK4duo\MK4duo-master\MK4duo\MK4duo.ino:343:

sketch\src/feature/tmc/tmc.h: In member function 'void MKTMC::refresh_hybrid_thrs(float)':

sketch\src/feature/tmc/tmc.h:305:125: error: 'hybrid_thrs' is not a member of 'TMC2208Stepper'

         inline void refresh_hybrid_thrs(const float spmm) { TMC2208Stepper::TPWMTHRS(tmc_thrs(TMC2208Stepper::microsteps(), TMC2208Stepper::hybrid_thrs, spmm)); }

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

--------------------------------------------------------------
Configuration_Feature.h ...

#define  X_DRIVER_TYPE  TMC2208
#define  Y_DRIVER_TYPE  TMC2208
#define  Z_DRIVER_TYPE  TMC2208
#define X2_DRIVER_TYPE  A4988
#define Y2_DRIVER_TYPE  A4988
#define Z2_DRIVER_TYPE  TMC2208
#define Z3_DRIVER_TYPE  A4988
#define E0_DRIVER_TYPE  A4988
#define E1_DRIVER_TYPE  A4988
#define E2_DRIVER_TYPE  A4988
#define E3_DRIVER_TYPE  A4988
#define E4_DRIVER_TYPE  A4988
#define E5_DRIVER_TYPE  A4988

...

#define Z_TWO_STEPPER_DRIVERS

--------------------------------------------------------------
Configuration_Motor.h ...

#define TMC_DEBUG

#define X_STEALTHCHOP   true
#define X2_STEALTHCHOP  false
#define Y_STEALTHCHOP   true
#define Y2_STEALTHCHOP  false
#define Z_STEALTHCHOP   true
#define Z2_STEALTHCHOP  true
#define Z3_STEALTHCHOP  false
#define E0_STEALTHCHOP  false
#define E1_STEALTHCHOP  false
#define E2_STEALTHCHOP  false
#define E3_STEALTHCHOP  false
#define E4_STEALTHCHOP  false
#define E5_STEALTHCHOP  false

#define CHOPPER_TIMING CHOPPER_DEFAULT_24V

#define HYBRID_THRESHOLD
#define X_HYBRID_THRESHOLD      90
#define X2_HYBRID_THRESHOLD    100
#define Y_HYBRID_THRESHOLD      90
#define Y2_HYBRID_THRESHOLD    100
#define Z_HYBRID_THRESHOLD      20
#define Z2_HYBRID_THRESHOLD     20
#define Z3_HYBRID_THRESHOLD      2
#define E0_HYBRID_THRESHOLD     30
#define E1_HYBRID_THRESHOLD     30
#define E2_HYBRID_THRESHOLD     30
#define E3_HYBRID_THRESHOLD     30
#define E4_HYBRID_THRESHOLD     30
#define E5_HYBRID_THRESHOLD     30

#define X_SERIAL_TX_PIN    40
#define X_SERIAL_RX_PIN    63
#define X2_SERIAL_TX_PIN  NoPin
#define X2_SERIAL_RX_PIN  NoPin

#define Y_SERIAL_TX_PIN    59
#define Y_SERIAL_RX_PIN    64
#define Y2_SERIAL_TX_PIN  NoPin
#define Y2_SERIAL_RX_PIN  NoPin

#define Z_SERIAL_TX_PIN    44
#define Z_SERIAL_RX_PIN    66
#define Z2_SERIAL_TX_PIN   42
#define Z2_SERIAL_RX_PIN   65
#define Z3_SERIAL_TX_PIN  NoPin
#define Z3_SERIAL_RX_PIN  NoPin

#define E0_SERIAL_TX_PIN  NoPin
#define E0_SERIAL_RX_PIN  NoPin
#define E1_SERIAL_TX_PIN  NoPin
#define E1_SERIAL_RX_PIN  NoPin
#define E2_SERIAL_TX_PIN  NoPin
#define E2_SERIAL_RX_PIN  NoPin
#define E3_SERIAL_TX_PIN  NoPin
#define E3_SERIAL_RX_PIN  NoPin
#define E4_SERIAL_TX_PIN  NoPin
#define E4_SERIAL_RX_PIN  NoPin
#define E5_SERIAL_TX_PIN  NoPin
#define E5_SERIAL_RX_PIN  NoPin

--------------------------------------------------------------

closed 2019-05-23 by doggyfan

iosonopersia, 2019-05-23:
Hi @doggyfan , is everything ok now?

doggyfan, 2019-05-23:
I don't know to be honest @iosonopersia, because I got no help with the problem I moved back to Marlin I'm afraid.