MKFirmware/MK4duo #712: Ultratronics V1.0 SD Card Read Error

thpparent, 2019-05-09:
With default parameter (unzip+BOARD set 1705) and #SDSUPPORT defined (to enable SD) 
- SD Card insert is detected (removed too)
- No SD Card is displayed

With #define SD_QUARTER_SPEED
- SD Card is recognised ONLY after REBOOT
- Print from SD Card appears in menu, content is correctly displayed (Directory) BUT printing gives ERROR : SD Read Error

M4duo V 4.3.9

Tried several SD card type size brand, no improvement
Tried with power supply or powered by USBP, no improvement

1705 pins .H extract (full attached)
//###MISC
#define ORIG_PS_ON_PIN             32
#define ORIG_BEEPER_PIN            27
#define LED_PIN                    13
#define SDPOWER_PIN                NoPin
#define SD_DETECT_PIN              60
#define SDSS                       59 //59
#define KILL_PIN                   NoPin
#define DEBUG_PIN                  NoPin
#define SUICIDE_PIN                NoPin

Thank you so much for your help

[seubm.zip](https://github.com/MKFirmware/MK4duo/files/3160929/seubm.zip)

thpparent, 2019-05-09:
After deeper investigation...

It is directly related to the version 4.3.9.

I have downloaded and configured the version 4.3.8 and it works like a charm

Thanks

MagoKimbra, 2019-05-09:
In 4.3.9 there is the new SDFAT library. I have an ultratronics and with my SD I have no problem. Now there is also the command to format the SD directly from firmware, try to format it with the appropriate command.

closed 2019-05-26 by MagoKimbra

iosonopersia, 2019-05-27:
Related to #702 . Problem fixed by @MagoKimbra