MKFirmware/MK4duo #713: adding the fystec card F6

musicana2018, 2019-05-11:
**Please, follow this template when opening a new issue: this will help us a lot in the process of helping you.**

### Issue description
hello
i have tried to add the new fystec f6 card to the firmware following their wiki for marlin but without any success and when i have made the configuration file i used custom board.
but i always get error compiling files.
thanks
here is the link for thier wiki page: https://wiki.fysetc.com/F6_V1.3/#step2-fysetc-arduino-plugin

### Compile errors
[error.txt](https://github.com/MKFirmware/MK4duo/files/3169188/error.txt)

### Firmware version
last version .
[board.txt](https://github.com/MKFirmware/MK4duo/files/3169191/board.txt)
[configuration_overall.txt](https://github.com/MKFirmware/MK4duo/files/3169192/configuration_overall.txt)

_If needed, you'll be asked for other settings you modified._

**Please be patient, we'll be trying to help you as soon as possible.**

iosonopersia, 2019-05-11:
Hi @musicana2018 ,
you get those errors because MK4duo does not have the pin definition file for that particular board. If this board is actually used by MK4duo users, we could add it. @MagoKimbra

musicana2018, 2019-05-11:
hi
thanks for replying so soon 
as i mentioned earlier
[pins_FYSETC_F6_131.txt](https://github.com/MKFirmware/MK4duo/files/3169278/pins_FYSETC_F6_131.txt)
 that this card is some sort of new one and i don't know if there some people use it or not, and for the pins number in theirs wiki page they have a pins file that need to be added in the firmware main page (marlin firmware) but when i add it their it seems that the firmware is not able to read it.
thanks again for your help

MagoKimbra, 2019-05-12:
Yes, that's fine, but if you see the other files on the other tabs you'll notice that the pin names are ORIG_
So you have to change all the names as well as the other tabs otherwise they won't be read by the firmware ...

musicana2018, 2019-05-13:
good morning
thanks for your help changing the pins file worked great but i am not able to compile the file as it tell me that i have to set a valid motherboard even i changed the board type in the file configuration overall to the good borad
[error.txt](https://github.com/MKFirmware/MK4duo/files/3173890/error.txt)
thank you

MagoKimbra, 2019-05-13:
In the file 541.h you must put #define KNOW_BOARD

musicana2018, 2019-05-13:
hello and thanks for your help
now all the error that i have before is fixed but i have ne error appeared and can't find the original file that i need to modify so it can work
[error.txt](https://github.com/MKFirmware/MK4duo/files/3174540/error.txt)

thanks for your help again

MagoKimbra, 2019-05-13:
Error on display u8g, Which display did you set?

musicana2018, 2019-05-13:
It is the full graphic reprap display that use the u8 library

MagoKimbra, 2019-05-13:
Ok i have full graphic display and 8 bit card... Check the pin in your file 541

musicana2018, 2019-05-13:
hello
i want to thank you again for your help
i have checked the pins in my 541.H file and they are correct. but it seems that the u8g library is not good as all the error message go back to that library.
[error.txt](https://github.com/MKFirmware/MK4duo/files/3174830/error.txt).
tomorrow i will reconfigure marlin from scratch and made all the change and see if is gonna to work
thanks

MagoKimbra, 2019-05-13:
Send be all file your modificated, board.h, 541.h a try and check the error..

musicana2018, 2019-05-13:
ok thanks again for your help
here you find all the files that i modified

[541.txt](https://github.com/MKFirmware/MK4duo/files/3174904/541.txt)
[Boards.txt](https://github.com/MKFirmware/MK4duo/files/3174905/Boards.txt)
[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/3174906/Configuration_Overall.txt)

MagoKimbra, 2019-05-13:
I try compiled with processor MEGA2560 i have not error..

musicana2018, 2019-05-13:
ok
thank you now it is working i think
i have set in the board manager to use the fystec compiler as they use their own board name on the arduino ide but since i changed it to mega2560 it worked great.
hope it will be able to uplaod to the card.

but thank you again for your help

closed 2019-05-26 by MagoKimbra