MKFirmware/MK4duo #717: Error compiling for board Arduino/Genuino Mega or Mega 2560. when #define TEMP_SENSOR_COOLER

gearsawe, 2019-05-22:
### Issue description
Setting up Rumba board. get 100s of warning and ending with error after setting #define TEMP_SENSOR_COOLER    2 in Configuration_Temperature.h

### Compile errors
only post a small section of the 5000+ lines of warnings which may not be relevant

Arduino: 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
In file included from sketch\MK4duo.h:106:0,
                 from C:\Users\gearsawe\Documents\Arduino\MK4duo-master\MK4duo\MK4duo.ino:344:
sketch\src/inc/conditionals_post.h:436:0: warning: "X_STALL_SENSITIVITY" redefined
   #define X_STALL_SENSITIVITY 0
 ^
In file included from sketch\MK4duo.h:92:0,
                 from C:\Users\gearsawe\Documents\Arduino\MK4duo-master\MK4duo\MK4duo.ino:344:
sketch\Configuration_Motor_Driver.h:90:0: note: this is the location of the previous definition
 #define X_STALL_SENSITIVITY 8
In file included from sketch\src/platform/HAL_AVR/HAL.h:98:0,
                 from sketch\src/platform/platform.h:49,
                 from sketch\MK4duo.h:112,
                 from C:\Users\gearsawe\Documents\Arduino\MK4duo-master\MK4duo\MK4duo.ino:344:
........
sketch\src/platform/HAL_AVR/fastio.h:1069:0: warning: "SCK" redefined
   #define SCK         DIO52
 ^
In file included from sketch\src/platform/platform.h:48:0,
                 from sketch\MK4duo.h:112,
                 from C:\Users\gearsawe\Documents\Arduino\MK4duo-master\MK4duo\MK4duo.ino:344:
sketch\src/platform/HAL_AVR/spi_pins.h:61:0: note: this is the location of the previous definition
 #define SCK   SCK_PIN
 ^
.....
 ^
In file included from c:\users\gearsawe\appdata\local\temp\arduino_build_760566\sketch\mk4duo.h:106:0,
                 from sketch\src\feature\babystep\babystep.cpp:29:
c:\users\gearsawe\appdata\local\temp\arduino_build_760566\sketch\src/inc/conditionals_post.h:437:0: warning: "Y_STALL_SENSITIVITY" redefined
   #define Y_STALL_SENSITIVITY 0
 ^
In file included from c:\users\gearsawe\appdata\local\temp\arduino_build_760566\sketch\mk4duo.h:92:0,
                 from sketch\src\feature\babystep\babystep.cpp:29:
c:\users\gearsawe\appdata\local\temp\arduino_build_760566\sketch\Configuration_Motor_Driver.h:91:0: note: this is the location of the previous definition
 #define Y_STALL_SENSITIVITY 8
.....
In file included from c:\users\gearsawe\appdata\local\temp\arduino_build_760566\sketch\src/platform/HAL_AVR/HAL.h:98:0,
                 from c:\users\gearsawe\appdata\local\temp\arduino_build_760566\sketch\src/platform/platform.h:49,
                 from c:\users\gearsawe\appdata\local\temp\arduino_build_760566\sketch\mk4duo.h:112,
                 from sketch\src\platform\HAL_DUE\HAL_timers.cpp:53:
c:\users\gearsawe\appdata\local\temp\arduino_build_760566\sketch\src/platform/HAL_AVR/fastio.h:1072:0: warning: "SS" redefined
   #define SS          DIO53
 ^
In file included from c:\users\gearsawe\appdata\local\temp\arduino_build_760566\sketch\src/platform/platform.h:48:0,
                 from c:\users\gearsawe\appdata\local\temp\arduino_build_760566\sketch\mk4duo.h:112,
                 from sketch\src\platform\HAL_DUE\HAL_timers.cpp:53:
c:\users\gearsawe\appdata\local\temp\arduino_build_760566\sketch\src/platform/HAL_AVR/spi_pins.h:64:0: note: this is the location of the previous definition
 #define SS    SS_PIN
 ^
**Error compiling for board Arduino/Genuino Mega or Mega 2560.**

Most of the issues are with the defines of fastio.h and others in src/platform/HAL_AVR/
or
Configuration_Motor_Driver.h and conditionals_post.h

### Firmware version
#define FIRMWARE_NAME             "MK4duo"
#define SHORT_BUILD_VERSION       "4.3.9"
#define FIRMWARE_REVISION         "13052019"

### My setup
Configuration_Basic.h
#define MOTHERBOARD 80
#define EXTRUDERS 2
#define DRIVER_EXTRUDERS 2

Configuration_Features.h
#define  X_DRIVER_TYPE  DRV8825 
#define  Y_DRIVER_TYPE  DRV8825 
#define  Z_DRIVER_TYPE  DRV8825 
#define E0_DRIVER_TYPE  DRV8825 
#define E1_DRIVER_TYPE  DRV8825 
#define SDSUPPORT
boards\80.h
#define ORIG_HEATER_HE2_PIN        NoPin// 6
#define ORIG_HEATER_COOLER_PIN     6// NoPin
#define ORIG_TEMP_HE2_PIN          NoPin// 13
#define ORIG_TEMP_COOLER_PIN       13// NoPin
#define ORIG_FAN0_PIN              8// 7
#define ORIG_FAN1_PIN              7// 8

Configuration_Temperature.h
#define TEMP_SENSOR_HE0       2
#define TEMP_SENSOR_HE1       2
#define TEMP_SENSOR_BED0      2
**#define TEMP_SENSOR_COOLER    2**

Configuration_LCD.h
#define MKS_MINI_12864
#define DOGLCD_CS        38//added
#define DOGLCD_A0        41//added
#define ENCODER_PULSES_PER_STEP 4
#define ENCODER_10X_STEPS_PER_SEC 30
//#define ULTIPANEL_FEEDMULTIPLY

MagoKimbra, 2019-05-22:
You must setting arduino ide with warning none.
OK but where is the problem?
Why you chenge the pin in 80.h and not in configuration_pins?
Why you write #define MOTHERBOARD 80 and not #define MOTHERBOARD BOARD_RUMBA?

gearsawe, 2019-05-23:
Updated the post. only defined necessary setting with a new download. Get a compile error after setting 
TEMP_SENSOR_COOLER 1 in Configuration_Temperature.h
Hope this help

MagoKimbra, 2019-05-23:
Delete thew //ORIG_.....
Because i use TEMP_COOLER_PIN in onther #define in firmware, and the IDE compilator use it but with // and cause error...

#define COOLER_ANALOG_CHANNEL     CHAMBER3_ANALOG_COMMA TEMP_COOLER_PIN

This line becomes #define COOLER_ANALOG_CHANNEL 
 ,ORIG_TEMP_HE2_PIN//ORIG_TEMP_COOLER_PIN

gearsawe, 2019-05-23:
Thanks! Hope this is right. Still working on the design. this firmware will make it possible. Not completely understanding why I have force the comma.
in conditional_post.h change
#define COOLER_ANALOG_CHANNEL     CHAMBER3_ANALOG_COMMA TEMP_COOLER_PIN
to
#define COOLER_ANALOG_CHANNEL     , TEMP_COOLER_PIN

and
in 80.h (Rumba Board)
#define ORIG_HEATER_HE2_PIN         6
#define ORIG_HEATER_COOLER_PIN      NoPin6
#define ORIG_TEMP_HE2_PIN          13
#define ORIG_TEMP_COOLER_PIN       NoPin
to
#define ORIG_HEATER_HE2_PIN        NoPin
#define ORIG_HEATER_COOLER_PIN      6
#define ORIG_TEMP_HE2_PIN          NoPin
#define ORIG_TEMP_COOLER_PIN       13

closed 2019-05-23 by gearsawe

MagoKimbra, 2019-05-23:
because it depends on the previous one, if CHAMBER3_ANALOG_COMMA is ',' or CHAMBER2_ANALOG_COMMA which in turn depends on the previous ...