MKFirmware/MK4duo #719: correct this

iluky, 2019-05-26:
https://github.com/MKFirmware/MK4duo/blob/e8eec3bbf1d7bd65833d3e06a7cb2eec98e520e1/MK4duo/src/feature/advanced_pause/advanced_pause.cpp#L544

it takes the negative sign

MagoKimbra, 2019-05-26:
No because unload_length just is negative number!!!

iluky, 2019-05-26:
it only works if I put the negative sign ... but maybe the problem is where you say

iluky, 2019-05-26:
:\Users\iluky\Dropbox\Firmware Taz i5L\05052019\MK4duo-master\MK4duo\src\commands\gcode\feature\m600.h:
  107      #if HAS_MMU2
  108        // For MMU2 reset retract and load/unload values so they don't mess with MMU filament handling
  109:       constexpr float unload_length     = 0.5f,
  110                        slow_load_length  = 0.0f,
  111                        fast_load_length  = 0.0f;
  112      #else
  113        // Unload filament
  114:       const float unload_length = ABS(parser.seen('U')  ? parser.value_axis_units(E_AXIS)

maybe the problem is there?

MagoKimbra, 2019-05-26:
But it is already like that, in the last fix I put !!

iluky, 2019-05-26:
yes now it's ok! thanks so much

closed 2019-05-26 by iluky