MKFirmware/MK4duo #720: Use the AC712 sensor on the 250Vac power supply.

Djac1974, 2019-06-01:
Can I use the AC712-5A directly on the 250Vac, the basic configuration and made on the 12Vcc but we don't measure the totality of the current used by the machine. (Heating bed: 24Vcc, and the voltages 3.3Vcc and 5Vcc, consumption of the screen Nextion: 5VDC, etc ...)

If yes what are the parameters to change
    // # define POWER_CONSUMPTION
    #define POWER_VOLTAGE 12.00
    #define POWER_SENSITIVITY 0.066
    #define POWER_OFFSET 0.005
    #define POWER_ZERO 2.500
    #define POWER_ERROR 0.0
    #define POWER_EFFICIENCY 100.0
    // # define POWER_CONSUMPTION_LCD_DISPLAY

Thank you in advance and congratulations MagoKimbra for version 4.3.9 which works super well.

MagoKimbra, 2019-06-01:
You need to activate # define POWER_CONSUMPTION and # define POWER_CONSUMPTION_LCD_DISPLAY to read on the display.
Now the problem that the sensor gives on the pin the current absorbed on 250 V, so it cannot give you the single absorption, but the total on the 250V at the head of the power supply.

Djac1974, 2019-06-01:
Thank you Magokimbra for your answer,
I need more information, can confirm that this configuration is good.
    # define POWER_CONSUMPTION 
    #define POWER_VOLTAGE **250**    //*/12.00 
    #define POWER_SENSITIVITY **0.185**    datashett AC712     //*/0,066 
    #define POWER_OFFSET 0,005   using **"M70 Z"**
    #define POWER_ZERO 2,500 
    #define POWER_ERROR   0.0
    #define POWER_EFFICIENCY 100,0 
    # define POWER_CONSUMPTION_LCD_DISPLAY

MagoKimbra, 2019-06-01:
This line is error
#define POWER_SENSITIVITY 0.185 datashett AC712 ///0,066
#define POWER_OFFSET 0,005 using "M70 Z"

write this
#define POWER_SENSITIVITY 0.185 //datashett AC712 0,066
#define POWER_OFFSET 0,005

doggyfan, 2019-06-01:
DO NOT use this sensor between the 250V AC mains and your printer board **without** proper opto (or transformer) isolation in between the two !!!

Djac1974, 2019-06-14:
Hello,

I receved AC712-5A
I test power consumption; I have problem during check.

**My configuration overhall:**

#define POWER_CONSUMPTION  //*p/
#define POWER_VOLTAGE      250  //*p/12.00    //(V) The power supply OUT voltage
#define POWER_SENSITIVITY   0.185  //*p/0.066   //(V/A) How much increase V for 1A of increase
#define POWER_OFFSET        0.005   //(A) Help to get 0A when no load is connected.
#define POWER_ZERO          2.500   //(V) The /\V coming out from the sensor when no current flow.
#define POWER_ERROR         0.0     //(%) Ammortize measure error.
#define POWER_EFFICIENCY  100.0     //(%) The power efficency of the power supply
//#define POWER_CONSUMPTION_LCD_DISPLAY  //*p/

**the status of check**

ATTENTION : la bibliothèque TMCStepper prétend être exécutable sur la (ou les) architecture(s) (avr) et peut être incompatible avec votre carte actuelle qui s'exécute sur (sam).
In file included from sketch\src\core\printcounter\../../../src/platform/platform.h:33:0,

                 from sketch\src\core\printcounter\../../../MK4duo.h:112,

                 from sketch\src\core\printcounter\printcounter.cpp:23:

sketch\src\core\printcounter\printcounter.cpp: In static member function 'static void PrintCounter::showStats()':

sketch\src\core\printcounter\../../../src/platform/common/communication.h:108:67: error: call of overloaded 'print(const char [17], uint32_t&)' is ambiguous

 #define SERIAL_VAL(val, ...)        Com::print(val, ## __VA_ARGS__)

                                                                   ^

sketch\src\core\printcounter\../../../src/platform/common/communication.h:120:59: note: in expansion of macro 'SERIAL_VAL'

 #define SERIAL_MV(msg,val, ...)     do{ SERIAL_MSG(msg);  SERIAL_VAL(val, ## __VA_ARGS__);                    }while(0)

                                                           ^

sketch\src\core\printcounter\printcounter.cpp:136:5: note: in expansion of macro 'SERIAL_MV'

     SERIAL_MV(CFG, "Watt/h consumed:", data.consumptionHour);

     ^

sketch\src\core\printcounter\../../../src/platform/common/communication.h:108:67: note: candidates are:

 #define SERIAL_VAL(val, ...)        Com::print(val, ## __VA_ARGS__)

                                                                   ^

sketch\src\core\printcounter\../../../src/platform/common/communication.h:120:59: note: in expansion of macro 'SERIAL_VAL'

 #define SERIAL_MV(msg,val, ...)     do{ SERIAL_MSG(msg);  SERIAL_VAL(val, ## __VA_ARGS__);                    }while(0)

                                                           ^

sketch\src\core\printcounter\printcounter.cpp:136:5: note: in expansion of macro 'SERIAL_MV'

     SERIAL_MV(CFG, "Watt/h consumed:", data.consumptionHour);

     ^

sketch\src\core\printcounter\../../../src/platform/common/communication.h:75:17: note: static void Com::print(char, int) <near match>

     static void print(char, int=BYTE);

                 ^

sketch\src\core\printcounter\../../../src/platform/common/communication.h:75:17: note:   no known conversion for argument 1 from 'const char [17]' to 'char'

sketch\src\core\printcounter\../../../src/platform/common/communication.h:76:17: note: static void Com::print(unsigned char, int) <near match>

     static void print(unsigned char, int=DEC);

                 ^

sketch\src\core\printcounter\../../../src/platform/common/communication.h:76:17: note:   no known conversion for argument 1 from 'const char [17]' to 'unsigned char'

sketch\src\core\printcounter\../../../src/platform/common/communication.h:77:17: note: static void Com::print(int, int) <near match>

     static void print(int, int=DEC);

                 ^

sketch\src\core\printcounter\../../../src/platform/common/communication.h:77:17: note:   no known conversion for argument 1 from 'const char [17]' to 'int'

sketch\src\core\printcounter\../../../src/platform/common/communication.h:78:17: note: static void Com::print(unsigned int, int) <near match>

     static void print(unsigned int, int=DEC);

                 ^

sketch\src\core\printcounter\../../../src/platform/common/communication.h:78:17: note:   no known conversion for argument 1 from 'const char [17]' to 'unsigned int'

sketch\src\core\printcounter\../../../src/platform/common/communication.h:79:17: note: static void Com::print(long int, int) <near match>

     static void print(long, int=DEC);

                 ^

sketch\src\core\printcounter\../../../src/platform/common/communication.h:79:17: note:   no known conversion for argument 1 from 'const char [17]' to 'long int'

sketch\src\core\printcounter\../../../src/platform/common/communication.h:80:17: note: static void Com::print(long unsigned int, int) <near match>

     static void print(unsigned long, int=DEC);

                 ^

sketch\src\core\printcounter\../../../src/platform/common/communication.h:80:17: note:   no known conversion for argument 1 from 'const char [17]' to 'long unsigned int'

exit status 1

**Advance thank you for your help...**

commit ecb2d55b (2019-06-16): Fix #720
closed 2019-06-16 by MagoKimbra