MKFirmware/MK4duo #725: autobed leveling not working with bl touch and z offset too

musicana2018, 2019-06-19:
goodmorning

for me the bl touch doesn't work at all ( keep reseting the bl touch all the time). and the few time that i made it work it probe fine with G29 but when i put the ZOFFSET  it seems it has no change over the distance of the nozzle of the bed. no matter what ZOFFSET i put it always keep the same distance.

MagoKimbra, 2019-06-19:
If it doesn't work, change what I have to tell you.
How do you change the Z_PROBE_OFFSET? What does the same distance always mean? When you do G1 Z0 does the distance between the nozzle and the plane change?

musicana2018, 2019-06-19:
good morning
thanks for replying
for of all to answer your question: i set the z offset by the M851 command and then i save it by M500 and apply it by M501.
but to let you know i figured it out and now it is working great. it seems that when you leave the z min in place and set the probe on the z max pin it never be able to work.
when i changed the probe to z min both for homing and for probing it is working great.
but my apologies i have just one other question left:
i set the fans for the extruder on autopilot via software, for the first fan it is working great as it kick at 50C
but my problem is that the blower for the hot end also work together with fan and never want to stop.
and could you please tell me if is there anyway to have an RGBW menu on the screen

thanks for your great help

MagoKimbra, 2019-06-19:
Ok for the probe part.
Fan problem. Do you have 2 fans?
The first fan is the 0 to cool the piece in PLA, it should not be used for autofan, because it is the one that is commonly used by all slicers.
Then give the command M106 P0 and write what the firmware writes.
Then give the command M106 P1 and do the same.
Do you have RGB LEDs for illuminating the printer? Yes there is the menu in the display.

musicana2018, 2019-06-19:
ok thanks for your reply
for the fan part i will test soon after i finish the long print and tell you the result
and about the rgb menu i have activated the rgbw led menu in the software but in the menu i am not able to see it.
could you please tell me where to find it
thanks

musicana2018, 2019-06-24:
good morning
thanks for helping me
i have teted the fan and the problem is fixed with your method it is working great, also the LED menu is there in the LCD menu.
my only problem is that hen i activate the TEMP_STAT_LEDS i am not able to compile and getting message error (please see attached file)
can you help me with that please.
thanks you
[config_overall.txt](https://github.com/MKFirmware/MK4duo/files/3321817/config_overall.txt)
[statled-error.txt](https://github.com/MKFirmware/MK4duo/files/3321818/statled-error.txt)