MKFirmware/MK4duo #728: 4.3.9 firmawre respond randomly with Nextion command

Ratisse, 2019-07-17:
Hi everybody

  I have upgraded the 4.3.8 firmware which worked fine with my MKS Gen 1.3 to the 4.3.9 firmware using same Configuration_Overall.h file in the online configurator .  But the only thing working is the hotend and bed temperature readings. Pushing several time one button sometimes works, for example touching about 10 times the text menu button on the Nextion display makes it appear; same for homing. It looks like responding very slowly after a period of time.

  Back to 4.3.8 firmware, everything  works fine, return to the 4.3.9 same problem. I have tried that many times with same results driving me mad. 

Thank you in advance for your help

[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/3401675/Configuration_Overall.txt)
[Configuration_Pins.txt](https://github.com/MKFirmware/MK4duo/files/3401676/Configuration_Pins.txt)

Ratisse, 2019-07-23:
Now works fine with latest updated firmware.

Thank you so much!!!

closed 2019-07-23 by Ratisse

MagoKimbra, 2019-07-23:
Ye sorry, with Arduino due i haven't problem with nextion. Yesterday i test Nextion with AVR and found the problem.