MKFirmware/MK4duo #729: Installation of a web camera

Micilera, 2019-07-18:
This is not an issue. It is more a design proposal.
We have a delta printer. With this architecture the piece is growing normally in the 'z' direction.
We would like to 'see' remotely the growing of the piece, so we think about installing a web camera  moving along the 'z' direction with a stepper motor (or in some other way). In this way be can follow remotely, but very close the activity of the printer.
The camera shouldn't be controlled in a very precise way, it should just follow the 'z' of the printer.
Some hint about this proposal? What do you think about? Could it be interesting to integrate the camera in the project? 
A moving camera is much more useful than a fix mounted camera.

Micilera, 2019-09-05:
An answer would be welcome. I accept any kind of answers, but please consider that an auxiliary axes that can be used by an user for a custom application would be definitively very useful, even if not very precise. Thanks.

GuttalBug, 2019-10-01:
I think is very interesting, open new capability. For example on delta printer a motorized bouchet solve the problem of purging the nozzle during the color change.