MKFirmware/MK4duo #731: CoreXY drop steps

Hans007a, 2019-07-31:
Anything is wrong with the last revision. My CoreXY printer (Due+ramps_fd v1) drops X-Y steps at different points. I posted my setting many times and I hope You have those, but when not I can post it again.

I will check the printer with an old revision.

Regards

Hans

MagoKimbra, 2019-08-04:
Sorry i do not understand your problem!!!
Explain  better.

Hans007a, 2019-08-04:
I attach a picture of the prints, look at the Z-axis. The first print is fully instable. I printed two test prints and You can see what is doing while printing. it looks like the printer is doing at a random height a random jump or forget to stops at the end of line...

[image]
[image]

Hans007a, 2019-08-04:
btw. I checked the whole mechanics, change the both belts and set the voltage on the drivers new

danandmary, 2019-10-04:
This looks like a classic case of stepper driver skipped steps.  Nothing to do with Firmware.