MKFirmware/MK4duo #732: Adding DHT11 sensor to 4.3.9 firmware freeze the board after firmware upload

Ratisse, 2019-08-08:
Hello

 The 4.3.9 firmware worked fine with my Nextion TFT4.3 and my MKS Gen 1.4 until I add the DHT sensor option to the online configurator and upload it to the board. When the board reboots the display is stuck on the text menu '"EEprom has changed push button". I tried to reload the working 4.3.9 firmware to init the Eeprom and reload the DHT one with same result, nothing works at all. The DHT is connected to the servo 4 connector D4. Do I miss something or is it a firmware issue? Sorry for my bad english, it's not my native language and thank you in advance.

My setup: 
[Configuration_Overall.h.txt](https://github.com/MKFirmware/MK4duo/files/3482759/Configuration_Overall.h.txt)
[Configuration_Pins.h.txt](https://github.com/MKFirmware/MK4duo/files/3482760/Configuration_Pins.h.txt)

MagoKimbra, 2019-08-08:
No no error, activating the DHT you add fields to the memory of the eeprom and therefore the one that has in memory is not valid, it is only warning you that it has loaded the default values and not those in eeprom, it is enough that you save with M500 and you have done , of course all the values changed and saved in eeprom you lost ...

Ratisse, 2019-08-08:
No way to save with M500, I can't do anything, the text menu is stuck. I tried to do this with Octoprint which returned a SD card init error. I also tried to desable the SD card on the online configuartor, reload the new firmware, always stuck on the text menu. Tried again to launch Octoprint which return a bad Eeprom size, very strange behaviour!!

Ratisse, 2019-08-09:
I managed to enter the M500 command with Pronterface, the "Eeprom changed" display has disappeared but now no Nextion command works. Octoprint works again and I could autotune the PID of hotend and bed and store to Eeprom. The Nextion displays "Autotune PID started" or the IP adress but nothing else, no temperature, no axis position etc.. I can home all axis with my phone app (Printoid), set the temperatures and read them on the phone but nothing with Nextion, so the board is not frozen. It looks like to be a Nextion-board communication issue

Ratisse, 2019-08-10:
After many firmware change, I've finally disabled the DHT option and reload the 08.08.2019 then the 10.08.2019 firmwares, the results are the same, the Nextion do not send the instructions to the board.
So I reloaded the last firmware before the 08.08.2019 one with the DHT manually activated and everything is working (except the DHT but it's another problem)

closed 2019-09-01 by Ratisse