MKFirmware/MK4duo #733: thermocouple support

maukcc, 2019-08-14:
If we us e -1 or -2 as temp sensor we get a 
TEMP_HE0_PIN was not declared
error when compiling.
In our pins file we have 
#define ORIG_HEATER_HE0_PIN        4
Is there anything we are missing?

MagoKimbra, 2019-08-14:
The thermocouples have specific pins on the board where there are SPI amplifiers. What card do you mount?
The pins are mapped to MAX6675_SS_PIN or MAX31855_SS0_PIN.

maukcc, 2019-08-14:
We have our own mainboard with a MEGA on it.

MagoKimbra, 2019-08-14:
Ok i fixed it in configuration_pins.
Re download...

maukcc, 2019-08-14:
Thanks, that worked.
But the readings are awful.
With the same setup and repetier or marlin I get a straight line for my temps.
Here I get a 10 degree dip every second or so. While the analog input signal is rock steady.

MagoKimbra, 2019-08-14:
The analogue reading is the same for the classic sensors that for the AD595 or AD596, what changes is the conversion formula.
When it reads the ADC pin it reads the digital conversion of the input value from 0 to 1024 (in decimal) then converts it to temperature with the rule:
raw * float (AD595_MAX) / float (AD_RANGE)) * ad595_gain + ad595_offset;

raw is the value read in decimal from 0 to 1024.
AD595_MAX is 500 which 5v for 100
AD_RANGE is 1024
the gain and the offset set with the M595 command is usually 1 for gain and 0 for offset.

If you dance 10 ° it means that the ADC reads a value that varies by 10 °.
If there were problems with the firmware you would have the same problem with the classic temperature sensors ...

closed 2019-08-14 by maukcc