MKFirmware/MK4duo #736: Nextion communication issue with 4.3.9 firmware

Ratisse, 2019-09-01:
My setup:
- Nextion 4.3 enhanced
- MKS Gen 1.4
- MK4duo 4.3.9 firmware

 After starting the printer, Nextion displays "Printer ready" and the IP address when connected to the Raspberry (used only to flash the firmware and tests with Octoprint or Printoid ). Nothing else works when pushing the Nextion keys, no move, no temperature display, no fan, etc...

  The firmware responds with Octoprint, just have a "SD card init failure" error. Disabling SD card do the same.

  The only 4.3.9 working firmaware is the "Short build version 22-07-2019", Each other one I tried with same configuration file are not working with Nextion. 4.3.8 firmware was working fine.

[Configuration_Overall.h.txt](https://github.com/MKFirmware/MK4duo/files/3563419/Configuration_Overall.h.txt)
[Configuration_Pins.h.txt](https://github.com/MKFirmware/MK4duo/files/3563420/Configuration_Pins.h.txt)

Hope you can help, thank you in advance

MagoKimbra, 2019-09-01:
Try now...

Ratisse, 2019-09-01:
Wow!! It's working now, thank you very much!!!

closed 2019-09-01 by Ratisse