MKFirmware/MK4duo #737: Bed heating issue while printing with 4.3.9 firmware

Ratisse, 2019-09-02:
My setup:
MKS Gen 1.4
Nextion 4.3 Enhanced
PID temp bed
Runout sensor set in the online configurator firmware but not connected (freeze the board)
No bed temp problem, runout sensor worked fine with 4.3.8 firmware

The issues:

- Bed autotunePID done and saved, with a bed temperature set to 55° the printer starts to print normally during several minutes with the bed temp stable at 55°, But suddenly after a while the temperature climbs up to 65° then stabilise and fall down to 57, stabilise again, fall to 53° rise again to 57°, stabilise to 55° during several minutes and etc... I know it's not a sensor or resistance problem, I have a LED mounted on the panel to check the heating activity

- Runout sensor connected, launching a print via SD card preheats bed and hotend, but when starting to home, the board crashes, the Nextion display is also frozen with the bed heating with no way to stop it except powering off (risk of fire). Disconnecting the sensor, the print is done normally with the issue above.

 Hope you have a solution, thanks in advance.

[Configuration_Pins.h.txt](https://github.com/MKFirmware/MK4duo/files/3566747/Configuration_Pins.h.txt)
[Configuration_Overall.h.txt](https://github.com/MKFirmware/MK4duo/files/3566748/Configuration_Overall.h.txt)

Ratisse, 2019-09-04:
Just tried your new firmware, now it's worse

- Runout sensor disconnected: when launching a print from Sd card, the print starts as soon without heating (temperature target displays 0° instead of 200°), Nextion displays "Printing file name"

- Runout sensor connected: same issue as above except Nextion displays "Hotend too cold"

MagoKimbra, 2019-09-05:
If you have the eeprom active, you must do pid and save it in eeprom, it tells you about the host log.
When you change the firmware, whatever it is, you must first redo all the controls by working with the host because it gives you all the information about things that are not right.
The filament runout can be active or inactive, there is the command and it is saved in eeprom if always active or not.
check the status.
As far as the problem that blocks the card is concerned, it cannot be a software problem, but a hardware one, check that the endstop of the runout is not short between + and gnd ...

Ratisse, 2019-09-05:
04.09.2019 firmware works better than the previous one (01.09.2019) which didn't block the card, just start printing without heating. Now it asks for pid before printing. PID of hotend and bed done and saved to eeprom, now the print starts after heating hotend and bed. 

But I still have the bed heating problem while printing. Asking 55°, the temperature is stable for several minutes then goes up to 65°, goes down and stabilize about 57°-59°, then goes down to 53°, sometimes is stable at 55° for a while etc..

What do you mean by "you must first redo all the controls by working with the host"? Is it a M502 followed by a M500 gcode commands? "Check the status" means M119 gcode command?

Maybe my MKS Gen 1.4 is not compatible with your 4.3.9 firmware The  4.3.8 one was working fine, I had no bed heating or runout sensor issue. I orderd and waiting for a MKS gen L v2 board with no TMC stepper wiring needed. Hope that one will work better even if not listed in your online configurator.

MagoKimbra, 2019-09-05:
Do you use pids for the bed? I don't use them on my 8-bit printer, and the bed doesn't move from 60 °.
The cards are all the same, the processor is the one that commands, what changes from one card to another is just the pin arrangement. But a card with the mega 2560 that is a ramps or an MKS or what you want does not change anything at all. In one the pin X_MIN_ENDSTOP is 12 and in another the 20 everything here.
With which firmware version does everything work properly ??

Ratisse, 2019-09-06:
Thank you for your clear explanations

Yes I use PID for the bed powered by 24V which goes easily to 105° and prints are much better.

4.3.8 Firmware_revision  20022019 was working fine but I can't use it anymore since I changed my stepper drivers to TMC 2208 and add DHT22. I've just tried to compil the last 4.3.8 version from your web configurator for testing purpose without success (Configuration_Pins.h:152: error: 'ORIG_HEATER_HE0_PIN' was not declared in this scope)

[Compil errors.txt](https://github.com/MKFirmware/MK4duo/files/3583378/Compil.errors.txt)

MagoKimbra, 2019-09-06:
Do you used configuration_overall.h (version 4.3.9) for download 4.3.8, the name ORIG_HEATER_HE0_PIN not used in 4.3.8 but use in 4.3.9.
Check the option selector in configuration and select Heather 0 normally used for hotend 0..

Ratisse, 2019-09-06:
Yes, I used configuration_overall.h (version 4.3.9). I will try what you say.

I've just noticed something else, after re-upload the 4.3.9 Nextion firmware I forgot to disconnect the runout sensor and tried to print a 20x20 cube then the Nextion displayed "MAX TEMP BED" and the hotend started to heat with no limits (Nextion displays: Hotend 26°-target 0°, Bed 26°-target 0°. Looks like a board crash with hotend everlast heating

MagoKimbra, 2019-09-06:
So I can't understand ... You have to give me some more indication, try again and check with the host what happens ...

Ratisse, 2019-09-06:
Just relaoded the 4.3.8 firmware with your indications and it works properly with runout sensor connected. PID done and saved for both hotend and bed, printing a 20x20 cube starts normally and the bed temp is very stable all the time (asking 55° stays at 55°). Just DHT not working 

What indications do you need? What do you mean by "check with the host" and how to do that?

Ratisse, 2019-10-07:
I have set the bed heating to bang bang which works a little better, now the temperature goes 3 degrees more than the asked temp on the first layer while printing and then goes down to normal temp

closed 2019-10-07 by Ratisse