MKFirmware/MK4duo #741: Wrong movement at delta autocal.

Arhigos, 2019-09-29:
I'm trying to setup mk4duo with my printer. I was able to do everything, only one issue that i have is when i'm doing autocalibration after homing it deploy pin, then moves to the middle and and start slowly moving UP instead of DOWN. and waiting for the pin to probe bed.

Other than that i can control and move everything in pronterface up and down, i can print models (if i remove g29 from begining in my g-code). I can deploy and stow sensor and it report triggered/not triggered correctly. 

[Configuration_Overall.txt](https://github.com/MKFirmware/MK4duo/files/3667356/Configuration_Overall.txt)
[Configuration_Pins.txt](https://github.com/MKFirmware/MK4duo/files/3667357/Configuration_Pins.txt)

### Firmware version
4.3.9

### My setup
Ramps 1.5
Atmega2560
Bltouch v3.1 (authentic from antclabs)

Looks like there was pretty similar issue that is already resolved https://github.com/MKFirmware/MK4duo/issues/420

MagoKimbra, 2019-09-29:
Chek with M119 if the probe is triggered or not when deploy.

Arhigos, 2019-09-29:
Yeah, as i said m119 works. It's not triggered when it deployed and triggered when stow.

Every time i send m119 it's changing probe position. For example i send m119 and it deploy, then i send m119 again and it stow.