MKFirmware/MK4duo #743: M218 does nothing

danandmary, 2019-10-02:
2 Extruders, 4.3.9
M218 T1 X5 Y5 Z5
Nothing happens
Check with M503
21:12:04.866 : Config:Hotend offset (unit): T<Tools> X<offset> Y<offset> Z<offset>:
21:12:04.870 : Config:  M218 T0 X0.000 Y0.000 Z0.000
21:12:04.870 : Config:Hotend offset (unit): T<Tools> X<offset> Y<offset> Z<offset>:
21:12:04.874 : Config:  M218 T1 X0.000 Y0.000 Z0.000
T0 - Nothing no movement
T1 - Nothing no movement

MagoKimbra, 2019-10-02:
Ok check this!

danandmary, 2019-10-03:
I found your commit after I got it working, but got compile error. ..." e/nozzle/nozzle.h:57:19: note:   candidate expects 1 argument, 0 provided."

 Here was my solution   Thanks for sorting this quickly.
/**
 * MK4duo Firmware for 3D Printer, Laser and CNC
 *
 * Based on Marlin, Sprinter and grbl
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 * Copyright (c) 2019 Alberto Cotronei @MagoKimbra
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program. If not, see <http://www.gnu.org/licenses/>.
 *
 */

/**
 * mcode
 *
 * Copyright (c) 2019 Alberto Cotronei @MagoKimbra
 */

 
#if HOTENDS > 1

#define CODE_M218

/**
 * M218 - Set hotend offset (in linear units)
 *
 *   T<tools>
 *   X<xoffset>
 *   Y<yoffset>
 *   Z<zoffset>
 */
inline void gcode_M218(void) {
	
   if (commands.get_target_driver(218)) return;
 //if (!commands.get_target_tool(218)) return;

  #if DISABLED(DISABLE_M503)
    // No arguments? Show M218 report.
    if (parser.seen_any()) {
      nozzle.print_M218(TARGET_HOTEND);
      return;
    }
  #endif

  if (parser.seen('X')) nozzle.data.hotend_offset[X_AXIS][TARGET_HOTEND] = parser.value_linear_units();
  if (parser.seen('Y')) nozzle.data.hotend_offset[Y_AXIS][TARGET_HOTEND] = parser.value_linear_units();
  if (parser.seen('Z')) nozzle.data.hotend_offset[Z_AXIS][TARGET_HOTEND] = parser.value_linear_units();
  

}  

#endif // HOTENDS > 1

MagoKimbra, 2019-10-03:
But did you just copy the M218.h file into your firmware, or all the firmware?
Because the function print_M218 in nozzle class no longer has any arguments so the print_M218 (TARGET_HOTEND) change is wrong.
I downloaded and compiled perfectly for more Hotend without problems.

closed 2019-10-22 by danandmary