MKFirmware/MK4duo #744: T0 and T1 servo positions mixed up

danandmary, 2019-10-03:
Using latest build 4.3.9
Have rocking hotend and using Dondolo dual extruder Bowden .
Enabled servos with positions 104 and 122
Also Dondolo e0 position 122 and E1 84
Firmware builds ok (one small unrelated error with z park which I overcame).
T0 moves to Dondolo position e0
T1 moves to Dondolo position E1
Everything fine.
Set offsets with M218
Run pid autotune on e0
Doesn't work for E1 so I manually copy the pid in Repetier firmware Configurator to E1
G29 moves servo to lower position , finishes and move servo to upper position.
So far so good.
Home printer, move out 50mm in X and Y

Here's the problem : 
T0 now moves to Dondolo servo position E1 
T1 now moves to Dondolo servo position E0 
Resetting printer, T0 and T1 are still reversed.
M502 then M500 
T0 is now back to normal 
T1 is back to normal
And the cycle continues. 
It appears as though the Dondolo servo positions get mixed up somehow in Eeprom
M503 doesnt report Dondolo servo positions so can only assume this is what is occuring as clearing Eeprom fixes the problem until I edit PIDs or offsets.

I'll post my configuration tomorrow morning.  (Sorry, on my phone right now).

MagoKimbra, 2019-10-03:
Ok.. Thank's

danandmary, 2019-10-04:
Here is my configuration

> /**
 * MK4duo Firmware for 3D Printer, Laser and CNC
 *
 * Based on Marlin, Sprinter and grbl
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 * Copyright (c) 2019 Alberto Cotronei @MagoKimbra
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program. If not, see <http://www.gnu.org/licenses/>.
 */

/**
 * Configuration_Overall.h
 * Here you can define all your custom settings and they will overwrite configurations in the main configuration files.
 */

/*******************************
 *   Firmware Version V4.3.9   *
 *******************************/

#define CONFIGURATION_OVERALL

/***********************
 * Configuration_Basic *
 ***********************/
#define SERIAL_PORT_1 0
#define BAUDRATE_1 250000
#define SERIAL_PORT_2 -2
#define BAUDRATE_2 250000
#define BLOCK_BUFFER_SIZE 32
#define MAX_CMD_SIZE 96
#define BUFSIZE 8
#define TX_BUFFER_SIZE 32
#define RX_BUFFER_SIZE 1024
#define SERIAL_XON_XOFF
//#define SERIAL_STATS_MAX_RX_QUEUED
//#define SERIAL_STATS_DROPPED_RX
#define STRING_CONFIG_AUTHOR "(none, default config)"
#define MACHINE_UUID "00000000-0000-0000-0000-000000000000"
#define KILL_METHOD 0
#define NO_TIMEOUTS 1000
//#define ADVANCED_OK
//#define EMERGENCY_PARSER
#define FASTER_GCODE_PARSER
#define FASTER_GCODE_EXECUTE
#define HOST_KEEPALIVE_FEATURE
#define DEFAULT_KEEPALIVE_INTERVAL 2
#define MOTHERBOARD BOARD_ULTRATRONICS
#define MECHANISM MECH_CARTESIAN
#define POWER_SUPPLY 0
//#define POWER_NAME "Generic"
#define PS_DEFAULT_OFF false
#define DELAY_AFTER_POWER_ON 5
#define POWER_TIMEOUT 30
#define EXTRUDERS 2
#define DRIVER_EXTRUDERS 2

/*****************************
[... 1743 lines cut ...]
"laserpwmpin": "ORIG_LASER_PWM_PIN",
"laserperipheralspin": "NoPin",
"laserperipheralsstatuspin": "NoPin",
"cncrouterpin": "NoPin",
"powerconsumptionpin": "NoPin",
"doorpin": "NoPin",
"powercheckpin": "NoPin",
"caselightpin": "NoPin",
"rgbledRpin": "NoPin",
"rgbledGpin": "NoPin",
"rgbledBpin": "NoPin",
"rgbledWpin": "NoPin",
"neopixelpin": "NoPin",
"dhtdatapin": "NoPin",
"tacho0pin": "NoPin",
"tacho1pin": "NoPin",
"tacho2pin": "NoPin",
"tacho3pin": "NoPin",
"tacho4pin": "NoPin",
"tacho5pin": "NoPin",
"END_DATA": 0,
"Zprobe": "1"
}
========== End configuration string ==========
*/

MagoKimbra, 2019-10-04:
Your configuration is not clear to me.
#define DONDOLO_SERVO_INDEX 1
#define DONDOLO_SERVOPOS_E0 0
#define DONDOLO_SERVOPOS_E1 0
But you only have one servo enabled, so the index must be 0.
You have 1 Servo the first servo has index 0.
You have 3 Servo the first is 0 the second is 1 the third is 2.
Then you have no set angle.
If you need a servo to do other set 2 servo where the servo 0 uses it for what you want and then the index of the DONDOLO is right at 1 and you will see the angles that can be set in M503.

danandmary, 2019-10-04:
Sorry that was an experiment to see if removing angles helps, I had infact set angles of 122 for e0 and 84 for E1.

Update: preliminary test indicate an issue with the use of SD Eeprom. Have changed to EEPROM FLASH.  Now seems to be working.  Will test more next week.

danandmary, 2019-10-07:
Unfortunately Changing to EEPROM FLASH did not fix the issue.  Still trying to figure the exact cause.  Ill report back if I find anything