MKFirmware/MK4duo #749: Tool Change Feedrate

danandmary, 2019-10-20:
Sorry to bother you guys, I know you're busy, but could someone explain how to adjust the tool change offset feedrate?  Change MAX Feedrate only appears to affect this up to a certain point at which is has no more effect.  The only way I can increase the speed at this stage is to adjust minimum travel feedrate, however this affects all travel movements.

MagoKimbra, 2019-10-20:
You can explain well what you need ...
Increase what?
Remember that the maximum feedrate of the axes limit all speeds to that speed, beyond which no thing can go further.

danandmary, 2019-10-20:
I am fully aware of the function of MAX Feedrate.  What I need to know, when  T1 is executed from command line, the toolhead moves to the position in the H1 X Y and Z Offset.  How do I adjust the speed of this movement?

MagoKimbra, 2019-10-20:
A ok with T1 F<feedrate you want>
Example T1 F3600 for 60 mm/s

danandmary, 2019-10-21:
Perfect.  That's exactly what i'm looking for.   Thank you.

danandmary, 2019-10-22:
Im afraid it doesnt appear to work.  T1 F100 - no change in speed  T1 F500 - not change in speed, T1 F6500 No Change in speed.   Do you have any ideas?  Max feedrate in eeprom is 10000

danandmary, 2019-10-24:
Can you confirm this?  tool change Feedrate not working for you too?    Do you know if it would be possible to hard code the feedrate for tool change?  I cannot figure out where in the firmware the toolchange feedrate is set.

MagoKimbra, 2019-10-24:
Excuse me, but I don't remember why the F command on T. had been eliminated. Maybe just a forgetfulness. However I put it back and then you can try again by downloading again.

danandmary, 2019-10-28:
Hi MK,  Sorry to keep bringing this up.  Having updated the code here https://github.com/MKFirmware/MK4duo/commit/0b16107a07aeb4fe90d613008bbfec77a533ffa9

I'm afraid to say it still doesn't work.  No change in feed rate.