MKFirmware/MK4duo #751: Compile error

marcubea1, 2019-10-25:
### Issue description
I used the online configurator and am having issues compiling

### Compile errors
Error was too long to post:

https://pastebin.com/TW4bKZDx

Before posting this issue, however, take a long breath and check this two things:
1. If you're not using the latest version of Arduino IDE, please try again with the latest one.
2. Always check if an error (or more than one) starts with `DEPENDENCY ERROR: _something wrong!_`: **DO NOT POST DEPENDENCY ERRORS**, please! They are messages which tell you what config option was set badly. Fix your configuration and try again. If you also get other types of error, please be sure to have fixed dependency errors **before** posting this issue!

### Firmware version 4.3.9
MK4duo's versions that have this issue.

### My setup
Config overall: https://pastebin.com/sy7k1ec2
Config pins: https://pastebin.com/rZpk1F4i

I apologise for the pastebin's the files wouldnt upload

Have you configured MK4duo manually? Then write here a list of the settings that may be related to the issue:
- SETTING_CHANGED_1    true
- SETTING_CHANGED_2    25
- SETTING_CHANGED_3    "blablabla"
- ...

_If needed, you'll be asked for other settings you modified._

**Please be patient, we'll be trying to help you as soon as possible.**

MagoKimbra, 2019-10-25:
Only warnings, disabled warning from arduino ide in menu. If it just disabled, put anather valor, close ide and reopen put none on warings ad compile...

marcubea1, 2019-10-25:
The sketch still won't compile for some reason, i apologise if i'm missing something simple

MagoKimbra, 2019-10-25:
Ok send me error, id you don't see warning in error report...

marcubea1, 2019-10-25:
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

MagoKimbra, 2019-10-25:
Only this errore in arduino log?

marcubea1, 2019-10-25:
I'll enable verbose and retry see if I can get more information

marcubea1, 2019-10-25:
In file included from C:\Users\bealm\AppData\Local\Temp\arduino_build_879995\sketch\src\commands\gcode/gcode.h:119:0,

                 from C:\Users\bealm\AppData\Local\Temp\arduino_build_879995\sketch\src\commands\commands.cpp:30:

C:\Users\bealm\AppData\Local\Temp\arduino_build_879995\sketch\src\commands\gcode/feature/m125.h: In function 'void gcode_M125()':

C:\Users\bealm\AppData\Local\Temp\arduino_build_879995\sketch\src\commands\gcode/feature/m125.h:57:36: error: 'struct nozzle_data_t' has no member named 'park_point'

   point_t park_point = nozzle.data.park_point;

                                    ^

C:\Users\bealm\AppData\Local\Temp\arduino_build_879995\sketch\src\commands\gcode/feature/m125.h:72:73: error: 'lcd_pause_show_message' was not declared in this scope

     lcd_pause_show_message(PAUSE_MESSAGE_PAUSING, PAUSE_MODE_PAUSE_PRINT);

                                                                         ^

C:\Users\bealm\AppData\Local\Temp\arduino_build_879995\sketch\src\commands\gcode/feature/m125.h:78:7: error: 'advancedpause' was not declared in this scope

   if (advancedpause.pause_print(retract, park_point, 0, show_lcd)) {

       ^

In file included from C:\Users\bealm\AppData\Local\Temp\arduino_build_879995\sketch\src\commands\gcode/gcode.h:219:0,

                 from C:\Users\bealm\AppData\Local\Temp\arduino_build_879995\sketch\src\commands\commands.cpp:30:

C:\Users\bealm\AppData\Local\Temp\arduino_build_879995\sketch\src\commands\gcode/sdcard/m24_m25.h: In function 'void gcode_M24()':

C:\Users\bealm\AppData\Local\Temp\arduino_build_879995\sketch\src\commands\gcode/sdcard/m24_m25.h:43:9: error: 'advancedpause' was not declared in this scope

     if (advancedpause.did_pause_print) {

         ^

"C:\\Users\\bealm\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\5.4.0-atmel3.6.1-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\SPI\\src" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\Wire\\src" "C:\\Users\\bealm\\AppData\\Local\\Temp\\arduino_build_879995\\sketch\\src\\core\\stepper\\stepper.cpp" -o "C:\\Users\\bealm\\AppData\\Local\\Temp\\arduino_build_879995\\sketch\\src\\core\\stepper\\stepper.cpp.o"
Multiple libraries were found for "SPI.h"
 Used: C:\Program
Multiple libraries were found for "Wire.h"
 Used: C:\Program
Using library SPI at version 1.0 in folder: C:\Program Files
Using library Wire at version 1.0 in folder: C:\Program Files
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

MagoKimbra, 2019-10-25:
OK you have enabled PARK_HEAD_ON_PAUSE, but this function requires NOZZLE PARK FEATURE...
If you have used the configurator, unfortunately the configurator remembers the enabled things even if they are hidden if the main function is disabled.
So if you have enabled the advanced pause and then the PARK_HEAD_ON_PAUSE and then you have disabled the advanced pause, PARK_HEAD_ON_PAUSE is enabled.

marcubea1, 2019-10-25:
Brilliant, all fixed thanks a lot for your help :)

MagoKimbra, 2019-10-25:
This is a problem that I will have to solve sooner or later on the configurator, but I have too many things to do that I always forget to do ...

closed 2019-10-26 by iosonopersia