MKFirmware/MK4duo #752: Compilation error

jmlcatarino, 2019-10-27:
**Please, follow this template when opening a new issue: this will help us a lot in the process of helping you.**

### Issue description
Downloaded new fW version and tried to upload to the board but got this strtange error, which didn't happen with older version. 
I have checked the Arduino lib, where it is supposed to be this file and it is there, although it has no file type, which I don't know if it's normal ou just a hidden file type. 

### Compile errors
arm-none-eabi-gcc: error: ../../../../../../Users/joaocatarino/Library/Arduino15/packages/arduino/hardware/sam/1.6.12/variants/arduino_due_x/libsam_sam3x8e_gcc_rel.a: No such file or directory

### Firmware version
MK4duo's versions that have this issue.
4.3.9 
### My setup
Laser, coreXY, tested same issue.
No settings were changed, just the version of the firmware was updated. 

_If needed, you'll be asked for other settings you modified._

**Please be patient, we'll be trying to help you as soon as possible.**

danandmary, 2019-11-12:
If you navigate to that folder, does the file exist?  What OS are you using?  Possible this has to do with https://github.com/MarlinFirmware/Marlin/issues/7967 ie long file names?  Tried compiling on Linux?

jmlcatarino, 2019-11-12:
@danandmary as stated on the issue description, the file is there. I was using Mac OS before the new update to Catalina. No I haven't, I don't use any Linux computer for compiling or for general use, tbh. 
I've seen your link but it just makes me go around and see other fixes, but it seems that it is something related to that.

danandmary, 2019-11-12:
I’m afraid you’re likely stuck if that is the issue.  I had a similar thing with Marlin 2.0. The only way I could fix the issue was to compile on a Linux platform that doesn’t have the max file length issue.  Apparently upgrading to newer version of IDE can fix this.  I’ve not had any success doing so, even with the latest beta build.