MKFirmware/MK4duo #756: Velleman Vertex Delta K8800 Board configuration

Bardje, 2019-11-12:
Currently the MK4duo firmware does not support the Velleman Vertex Delta K8800 Printer. Would it be possible to add this printer to the branch?

Velleman supplies a full schematic of how the ATMEGA2560 is connected to all the components. Most of the pinout is identical to the Velleman K8400 configuration which is already included in the MK4duo firmware, although the are some differences between the two.

I tried changing the pinout in the Pins.h file, but not all the pins can be configured there, for the LCD for example. So no luck for me. 

Thanks in advanced!

Here is the schematic of the Velleman Vertex Delta:
[K8800-schematic-V1.4.pdf](https://github.com/MKFirmware/MK4duo/files/3836199/K8800-schematic-V1.4.pdf)

Bardje, 2019-11-12:
I did some tinkering i think i have got it to work. I used the 79.h in the boards file as a template and made this to the pinout i got from the original firmware of the K8800.
Now the sketch will fully compile and i will try it on the 3D printer soon.

I named the K8800 file 80.h because this came after another product of theirs.
[80.txt](https://github.com/MKFirmware/MK4duo/files/3837205/80.txt)
Original K8800 firmware:
[pins_K8800.txt](https://github.com/MKFirmware/MK4duo/files/3837209/pins_K8800.txt)

iosonopersia, 2020-03-16:
Hi @Bardje , have you tried your pin configurations on your 3D printer yet?
If it does work, it can be added to the firmware.
Please let us know, it could be useful for other users too.